FreeCAD/FreeCAD #6041: Non-ASCII usernames in MediaWiki not working

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4299

* **Reporter:** kkremitzki
* **Date submitted:** 3/30/2020
* **FreeCAD version:** 
* **Category:** General
* **Status:** assigned
* **Tags:** 

# Original report text

For example, a username containing "andrë" does not work (can't be created.) Need to make sure the database encoding is correct.

# FreeCAD Info

```
OS: Debian GNU/Linux bullseye/sid
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4.
Build type: Release
Python version: 3.7.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.3.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Updated:** 2/6/2021

alwxkxk, 2022-05-12:
window 10 ,
FreeCAD V0.19
Program open crash if user name is chinese ,for Example:
It will crash if my user name is 'C:\Users\中文名称'.
It will work if my user name is 'C:\Users\englishname'

luzpaz, 2022-05-12:
@alwxkxk This ticket is for FreeCAD's documentation that is built on Mediawiki. It has nothing to do with FreeCAD itself. Also this is not a place to post bugs until they have been vetted on the forum. Please post to the forum first. BTW, The solution to your issue is updating to [FreeCAD v0.20dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds). I will erase this is a day or so, so you can still see this.