FreeCAD/FreeCAD #6043: 5 standard options are missing in the PartDesign Tree view context menu.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4308

* **Reporter:** Roy_043
* **Date submitted:** 4/10/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** 

# Original report text

5 standard options are missing in the PartDesign Tree view context menu:
"Toggle visibility" - "Select all instances"

# Steps to reproduce

Select an object in the tree and right-click.

# FreeCAD Info

```
OS: Windows 8.1 (6.3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20311 (Git)
Build type: Release
Branch: master
Hash: 915e551bbb7e3614bc804f1ae1f65027b5432b9f
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Dutch/Netherlands (nl_NL)

```

# Other bug information

* **Priority:** normal
* **Severity:** tweak
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Roy_043 2020-04-10 08:46
Forum topic:
https://forum.freecadweb.org/viewtopic.php?f=3&t=44948

maxwxyz, 2024-02-26:
@Roy-043 is this still relevant?

Roy-043, 2024-02-26:
Yes, this is still relevant.

closed 2024-03-28 by wwmayer