FreeCAD/FreeCAD #6045: Set "Maximum number of backup files" default to a larger number

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4313

* **Reporter:** freedman
* **Date submitted:** 4/12/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** feedback
* **Tags:** #pending-PR

# Original report text

Please set the default maximum number of backup files to a larger number, it is currently set at 1. I suggest a setting of 20, this is what it used to be set to. I can think of no down-side to doing this.

# Additional information

1 backup could easily be over-written if there is model problem. The user might restart FreeCAD a couple times trying to figure out what's wrong and could possibly lose their good version of the model.

I see the value of 1 set in 0.18.16 and in (just downloaded) 19.204.
Thanks

# Steps to reproduce

Look in Preferences-Document-Maximum number of backup files.

# FreeCAD Info

```
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16085 (Git)
Build type: Release
Branch: master
Hash: 6323069cbbda5ac71dff073f15cc632d8a1f3ea3
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** tweak
* **Category:** Feature
* **Updated:** 1/29/2022

# Discussion from Mantis ticket

### Comment by openBrain 2020-04-13 07:35
freedman : beyond following our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236), this kind of changes shall absolutely be discussed in the forum so we have a community consensus. Please open a topic and post link here. Thx

---

### Comment by openBrain 2020-04-14 08:09
[Forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=45232)

---

### Comment by Kunda1 2022-01-27 15:16
PR https://github.com/FreeCAD/FreeCAD/pull/5451

---

### Comment by Kunda1 2022-01-29 17:22
I think OP means minimum here...not maximum

adrianinsaval, 2022-04-04:
considering what is actually wanted is a higher "minimum" number of backup files, can someone change the title accordingly?

adrianinsaval, 2023-04-14:
I think both kunda and I had brain farts at the time, it is indeed the _maximum_ that was asked to be changed.

Was there any consensus regarding this? Is there really any reason to generate that much junk _by default_? I would even make the default 0, what other software out there makes a backup when you save a file? This kind of behavior should happen by user choice not by default.

I'm inclined to close this issue, it's been 3 years since the proposal and no action has been taken, changing a default value is not something that should take that long. Besides that if there isn't an argument that indicates that a majority of users would prefer/benefit from changing the default I don't think it should be changed, those who want this to be higher can just adjust the preference.

luzpaz, 2023-04-16:
> I think both kunda and I had brain farts at the time, it is indeed the maximum that was asked to be changed.

Lol, indeed!

> Was there any consensus regarding this?

I don't recall very well. 

> Is there really any reason to generate that much junk by default?

I think the junk is due to lack or placing the backups in their own respective directories. It's certainly not junk when you have a failure and miraculously can resurrect a crashed projects from where you left off because this feature was enabled.

adrianinsaval, 2023-04-16:
> I think the junk is due to lack or placing the backups in their own respective directories. It's certainly not junk when you have a failure and miraculously can resurrect a crashed projects from where you left off because this feature was enabled.

I think this is the big misconception that leads to this request... These ARE NOT recovery files, these backups are copies of the old file _when you save_. They will not help you recover anything from a crash, you'll get an even older version than you where working on. Recovery files are saved in the cache directory if I'm not mistaken, or in the temp files directory

yorikvanhavre, 2023-04-19:
> I think this is the big misconception that leads to this request... These ARE NOT recovery files, these backups are copies of the old file when you save.

Even so they are more than once very useful. Lots of other apps do the same, save a backup file on save, like AutoCAD or Blender. For me the current value is a sensible one. AutoCAD does 1 backup by default, Blender I think 2. It's easy to find the setting in the preferences, and if you haven't looked into the preferences yet, then the value of 1 seems perfect, so you "learn" about backup files when you start working, and that one might very well save your life ;) If you then want more than one, it's not hard to find where to change.

I would close the issue.

adrianinsaval, 2023-05-04:
agreed with yorik

closed 2023-05-04 by adrianinsaval