FreeCAD/FreeCAD #6046: Add visual orientation while creating datum plane

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4314

* **Reporter:** drmacro
* **Date submitted:** 4/14/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** planes

# Original report text

https://forum.freecadweb.org/viewtopic.php?f=3&t=45159

During creation of a datum plane provide visual feedback depicting the orientation of the X,Y,Z. For example while selecting 3 point plane.

Possibly toggle on for reference at later times, to clarify orientation when creating attached sketches.

# Steps to reproduce

Create a datum plane based on existing geometry.

# FreeCAD Info

```
OS: Ubuntu 19.10 (XFCE/ubuntustudio)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.7.5
Qt version: 5.12.4
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: All All**
* **Platform:** All
* **Updated:** 2/6/2021

adrianinsaval, 2022-07-01:
another discussion related to this
https://forum.freecadweb.org/viewtopic.php?f=8&t=69932
it would be nice to have this feature when editing any object's attachment, not just datum plane

DrD4ffy, 2024-02-07:
Just had the case where the plane was in the correct position, but front/rear or exact direction of the axis was unclear. Something like this ...

[image]

... would help.

maxwxyz, 2024-04-21:
Related: https://github.com/FreeCAD/FreeCAD/issues/12584

luzpaz, 2024-12-23:
@PaddleStroke could you kindly weigh-in ?

gryphonB-2, 2025-01-07:
I would like to add my strongest support for this issue.
At present the workflow demands that you create a datum plane, guessing what you want, then create a sketch on it to find out what you have, deleting it because it it isn't. You don't even know what side of the plane you will get, let alone the axis orientation. 
It's not that you can't get what you want, it's the unnecessary messing around in between.
Brighter people than me may not care, but it adds up to another thing to juggle instead of focussing on production: I don't read a book sideways & back to front, even though I can. 

Finally, I think it may affect the way Lofts are created, if some sketches are on the DP back & some on the front, since I just had a simple one fail, utterly criss-crossed between contours. 

Version info:
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.5.10
  * Curves 0.6.51
  * DynamicData 2.70.0
  * slic3r-tools
[image]
[Uploading BikeRack.FCStd.zip…]()

maxwxyz, 2025-01-07:
@kadet1090 FYI

pierreporte, 2025-06-13:
If it helps I’m not against it. However I’ve proposed new attachment panels, with a different UI for each kind of things to attach (point, line, plane, sketch) that would be easier to use and more powerful. Putting the axis now could be a quick fix that improves the situation until the better things is implemented.

maxwxyz, 2025-08-12:
There is a new placement indicator which can be activated in the view properties. Unfortunately it does not work for datum planes (but all other objects) @kadet1090 
If this would work on planes we could close this issue as duplicate of https://github.com/FreeCAD/FreeCAD/issues/22508