FreeCAD/FreeCAD #6047: FEM WB, WarpVector widget default position from Right to Left

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4315

* **Reporter:** fc_tofu
* **Date submitted:** 4/15/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** #ready-for-migration

# Original report text

!(https://forum.freecadweb.org/download/file.php?id=109887)
By default pref, NaviCube shows on TopRight corner of viewport, and WarpVector bar widget (FEM WB) cannot be moved.
NaviCube and WarpVector bar widget overlay at TopRight corner in every new installed FreeCAD.
A known workgound is to move NaviCube away to the left.
But right side of viewport is more "stable" and preferable. By "stable", I mean that cannot be affected by frequent ajustment of neighboring ComboView border position. Since NaviCube is far more used than WarpVector bar, so I hope the NaviCube can stay at right, and WarpVector bar moved to left as future default position.
!(https://forum.freecadweb.org/download/file.php?id=109888)
To prevent new conflict with other/potential widgets, a "Alighment = Left/Right" option can be added to view properties of WarpVector. 
!(https://forum.freecadweb.org/download/file.php?id=109889)

# Additional information

[Forum discussion](https://forum.freecadweb.org/viewtopic.php?f=8&t=45239&p=387898#p387898)

# Steps to reproduce

Open stock example file "FemCalculixCantilever3D.FCStd", then 
WarpVector bar widget appear on right of viewport with NaviCube overlayed on TopRight corner.

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20514 (Git)
Build type: Release
Branch: master
Hash: a05ffd499366ab12fa1b43f37db73f4f1e9c332d
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Chinese/China (zh_CN)

```

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/4/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2022-02-03 13:47
jnxd can you take a look at this ?

---

### Comment by Kunda1 2022-02-03 21:40
[quote=OpenBrain]
If I get it well, it's not trivial to move as the color bar actually lies in the "foreground" provided by Inventor View Provider : https://github.com/FreeCAD/FreeCAD/blob/8492cfb9eb49f6d882ceefb60440acd2a30fccfd/src/Mod/Fem/Gui/ViewProviderFemPostObject.cpp#L194
[/quote]

---

### Comment by jnxd 2022-02-04 19:49
Thanks for locating the exact place where the colorbar is implemented. Unfortunately there's no clear parameter in Gui::SoFCColorBar to just set it's position (I'd be happy to be proven wrong), so I'll need to dig in a bit.

While this bug itself should be fixed, I don't know how much of the post-processing we would want to handle locally within FreeCAD. There are other programs that are built for exactly this purpose (looking mainly at ParaView), and I believe it'd be kind of a duplication of effort to support more complex behavior.

luzpaz, 2023-02-26:
Wonder if @donovaly can take a look since he worked on #6935 ?

donovaly, 2023-02-26:
The FEM bar widget is a beast. We already spent a lot of time to get it working better but still many open issues. Therefore adding now a new property to move its position would not be optimal. (For example at the moment it even jumps around when you change the size of the main windows). The NaviCube can however easily be moved to any position you like:
left-click on it, keep the mouse button pressed and move the mouse