FreeCAD/FreeCAD #6049: Support incremental Gcode in postprocessors

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4320

* **Reporter:** sliptonic
* **Date submitted:** 4/21/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Some machine controls require gcode to be incrmental.  We currently don't have any postprocessors that work this way so there's no reference implementation.

https://forum.freecadweb.org/viewtopic.php?f=15&t=44554&p=380569&hilit=edm#p380569

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by sliptonic 2020-04-22 22:48
I really need a sample file and gcode to work with but here's an early attempt to try.

I made a 10x10 cube and profiled it without compensation.  Use the attached postprocessor with the following parameters 
--no-header --relative --axis-modal --line-numbers

The output looks like this:

N105 (begin preamble)
N110 G17 G54 G40 G49 G80 G91
N115  G91
N120 G21
N125 (begin operation: Fixture)
N130 (machine: not set, mm/min)
N135  G54
N140 (finish operation: Fixture)
N145 (begin operation: Spot Drill001)
N150 (machine: not set, mm/min)
N155  (Spot Drill001)
N160 M5
N165  M6 T2 
G43 H2
N170  M3 S0
N175 (finish operation: Spot Drill001)
N180 (begin operation: Contour)
N185 (machine: not set, mm/min)
N190  (Contour)
N195  (Uncompensated Tool Path)
N200  G0 Z+016000
N205  G0 X+005000 Y+005000
N210  G0 Z-002000
N215  G1 Z-014000
N220  G1 Y-010000
N225  G1 X-010000
N230  G1 Y+010000
N235  G1 X+010000
N240  G0 Z+016000
N245 (finish operation: Contour)
(begin postamble)
N250 M05
N255 G17 G54 G90 G80 G40
N260 M2

---

### Comment by sliptonic 2020-04-22 23:15
The attached post has two kinds of changes in it.  The changes to make the relative gcode are, I think, broadly applicable and should be included in the linuxcnc reference post processor.  

The changes to format the numbers in a specific way is unique to the Geg's EDM machine and should not be included.

luzpaz, 2022-02-11:
Heads up, there is a python file in the mantis ticket that didn't get migrated.

sliptonic, 2022-02-11:
[relative_post.py.gz](https://github.com/FreeCAD/FreeCAD/files/8049794/relative_post.py.gz)

Gegsite, 2022-02-21:
Hey!

Just find this.
I insert a 5mm notch's program.

AGIE EDM machine .GEO file is looks like this:

%
N00G01X+000001Y+000000G40M21
N10D01S01T01G43
N20G01X+000000Y+005325G44
N30G01X-002501Y+000000G40
N40G01X+005001Y+000000G44
N50G01X+000000Y-005325
N60G01Y-001000
N70G01X-000175Y+000000G40M21
N80G45M21
N90D02S02T02G43
N100G01X+000000Y+005325G44
N110G01X+000175Y+000000G40
N120G01X-005001Y+000000G44
N130G01X+000000Y-005325
N140G01Y-001000
N150G01X+004001Y+000000G40M21
N160G45M21M02

sliptonic, 2022-02-21:
Interesting.  
The D commands are unusual.  They appear to be setting up the cutting element and include a S word T word and G43.  
Is this a multi-spindle machine or perhaps controlling the top and bottom position of the EDM wire?

Gegsite, 2022-02-21:
If I know right, D is for diameter (from tech file) and so on. 
This (and many) EDM machine has a head that can tilt in U, and V, axis.Normally I dont use  it. Only  now will try to use it as I need to cut a mini lathe tool, of course the the tool is wedge from top to bottom (3°), but that needs tweak in the .JOB file, so no affect in the .GEO - gcode.
Actually the T01 and T02 is just for reversing the direction and fine cut the part, so it must inverse the offset.
I found a nice article about this:
http://www.cam-1.com/Agie_programming.htm

alromh87, 2023-11-01:
Just added a pull request that mostly covers this issue, tried to make it as flexible as possible to help as much people as possible, to support @Gegsite type of machine just set right arguments: 
`--line-numbers --relative --force-sign --fixed-length 6 --scale 1000 --precision 0  --two-digit-codes --no-comments --ignore-operations "Fixture,TC: WireEDM" --postamble "G45M21M02" --no-header --command-space "" --preamble ""`
 
Yields
```
N110
N120G91
N130G21
N140G01X-009410
N150G01Y+054910
N160G01X+018820
N170G01Y-109820
N180G01X-018820
N190G01Y+054910
N200G45M21M02
```

 I have verified it works in on an Pre 2000 BKDC WEDM machine that uses micrometers and has no Z axis, for reference I use:
`--relative --comments-character % --scale 1000 --precision 0 --endline ";" --modal --postamble "M02" --preamble "G90\nG92X0Y0" --omit-units --two-digit-codes --no-comments --ignore-operations "Fixture,TC: WireEDM"`

Yields:

```
%(Exported by FreeCAD)
%(Post Processor: wedm_post)
%(Output Time:2023-10-31 21:35:16.377564)
G90
G92X0Y0
G91
G01 X-9410;
Y54910;
X18820;
Y-109820;
X-18820;
Y54910;
M02
```

I could add incremental support on other postprocessors if so needed and also be happy to fully support @Gegsite machine if feedback and testing is provided

luzpaz, 2025-06-24:
cc @LarryWoestman has the postprocessor refactor impacted this feature request at all ? (or will it?)