FreeCAD/FreeCAD #6050: Import of any .dae (Collada)-file fails

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4323

* **Reporter:** cadandre
* **Date submitted:** 4/24/2020
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** feedback
* **Tags:** #post-to-forum, 0.18

# Original report text

According to
https://wiki.freecadweb.org/index.php?title=Extra_python_modules#pyCollada

FreeCAD on Windows should be able to open/import .dae-files
Quote:
'Windows

On Windows since 0.15 pycollada is included in both the FreeCAD release and developer builds so no additional steps are necessary.'

Nevertheless every import of .dae-files (Open as well as Import) fails (see screenshot attached). If somebody tells me where to find the errormessage at the bottom of the screenshot, I could provide it.

# Steps to reproduce

open/import any .dae-file

# FreeCAD Info

```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10**
* **Platform:** PC
* **Updated:** 2/7/2021

# Discussion from Mantis ticket

### Comment by openBrain 2020-04-24 22:38
cadandre : please address your issue according our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by opening a topic in our forum. Thx

---

### Comment by chennes 2021-02-07 21:41
Here's an example DAE file for testing.

---

### Comment by chennes 2021-02-07 22:08
I am able to load the example file under the following:
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23971 (Git)
Build type: Release
Branch: master
Hash: a3fb41502b079a86175e6a414d82956a48bab4e2
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

luzpaz, 2022-05-13:
I try to load the example file and see 
`16:52:08  pycollada not found, collada support is disabled.`  
Maybe we can provide a link on how to add install pycollada ?

Roy-043, 2023-07-16:
Without an actual file from cadandre this issue does not make sense. Closing it as stale.

luzpaz, 2023-07-19:
> Comment by chennes 2021-02-07 21:41
> Here's an example DAE file for testing.
> [ExampleCollada.dae.zip](https://github.com/FreeCAD/FreeCAD/files/8691184/ExampleCollada.dae.zip)

@Roy-043 :point_up:

Roy-043, 2023-07-20:
I truly do not understand why this was reopened.

IMO this bug report should have been rejected as incomplete before. The file chennes has provided contains a very simple mesh. It can be opened in V0.18 as well but this does not tell us much. We really need cadandre's file. But since this is an imported bug, cadandre is not notified about this discussion, so this file will remain illusive.

Informing Linux users on how to install dependencies is a different issue. If you want to address that then you should create a new issue.

Closing this again.

closed 2023-07-20 by Roy-043

yorikvanhavre, 2023-07-26:
Ideally we should have somewhere a kind of catch-all help page where users can land if they have import/export problems (maybe in the FAQ?). It is indeed usually hard to navigate and know which WB is responsible for what format