FreeCAD/FreeCAD #6051: Strange character after bracket, appear in expression input field for a constraint in sketcher

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4325

* **Reporter:** lc03
* **Date submitted:** 4/27/2020
* **FreeCAD version:** 
* **Category:** General
* **Status:** feedback
* **Tags:** #pending-forum

# Original report text

When I type in expression containing bracket in the expression editor a strange character appear at the end of the entered text.

# Steps to reproduce

1)	Create a new sketch
2)	Create 2 new line.
3)	Create a dimension constraint on the first line.
4)	Create a dimension constraint on the second line and then enter the “=” sign to edit expression.
5)	Type in the expression “Constraints[0]”
Result: just after you type in the last character (that is “]”) the strange character appear in the expression editor

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20738 (Git)
Build type: Release
Branch: master
Hash: 0c96f77184edc0a6747483b2ff1462fc6dae46ff
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2020-04-28 19:43
lc03 : Please address your issue according our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by first opening a topic on [our forum](https://forum.freecadweb.org/index.php) ("Help on using FreeCAD" subsection in English or "Forum français" in French) so community can vet the issue and narrow down the causes. Cross-link the topic with this ticket. Thx

---

### Comment by openBrain 2020-04-29 09:50
[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=12&t=45811)

---

### Comment by Kunda1 2020-12-08 01:25
Moved to Expressions category

luzpaz, 2022-05-25:
@0penBrain can you still reproduce?

0penBrain, 2022-05-26:
@luzpaz I never reproduced that because it seemed to be Windows+French specific (I'm lacking the Windows part 😄). I will bump the topic on the French subforum and report.

0penBrain, 2022-05-30:
What I understand from answer on the forum is that it's probably a Windows issue :
* Windows has a feature that when you hold `Alt` and enters a number with numpad, you can get special symbols
* Windows seems to interpret `AltGr` as `Ctrl+Alt`
So depending on how those virtual `Ctrl+Alt` are considered to be pressed then released, if you have used the numpad while holding `AltGr`, you can see weird symbols appearing.
There are things we can do, but looks to me like killing a fly with a bazooka...

luzpaz, 2023-12-14:
@PaddleStroke have you encountered this bug yet ? Any idea on how to auto-correct this phenomena so it doesn't present ?

PaddleStroke, 2023-12-14:
I have tried and cannot replicate. Even if I hold altGr while typing the '[0]' and use the numpad to type the 0, I don't get a special character to appear.

luzpaz, 2023-12-15:
Thanks paddle. 

Does anyone by chance know [mario52's](https://forum.freecad.org/memberlist.php?mode=viewprofile&u=1058) github account ? He's been the only one to reproduce this. 

I think we should close in the meantime.

closed 2023-12-15 by luzpaz