FreeCAD/FreeCAD #6052: Several "Scripted objects" examples don't work

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4328

* **Reporter:** tomkcook
* **Date submitted:** 4/28/2020
* **FreeCAD version:** 
* **Category:** General
* **Status:** new
* **Tags:** #post-to-forum

# Original report text

See the examples on [this wiki page](https://wiki.freecadweb.org/Scripted_objects).

The "Basic example" works as advertised.  But the "Other more complex example", when pasted directly into the FreeCAD python console, produces a couple of triangles, not the expected octahedron.

And when it is modified as suggested in the section "Making objects selectable", only a single red line is shown in the 3D display (which is then not selectable).

I have pasted my modified version of this code into the "Additional information" section because I couldn't make the file attachment work.

I have tried this on a couple of builds from source, most recently on commit bc107eacc8e020a779b2bbcb21559283ee440268, as well as the 0.18.3 that is distributed with Ubuntu 19.10 and the 0.18.4 that is distributed with Ubuntu 20.04.  None of them worked as expected, though there were minor variations in the results.

# Additional information

import FreeCAD, FreeCADGui, Part
import pivy
from pivy import coin

class Octahedron:
  def __init__(self, obj):
     "Add some custom properties to our box feature"
     obj.addProperty("App::PropertyLength","Length","Octahedron","Length of the octahedron").Length=1.0
     obj.addProperty("App::PropertyLength","Width","Octahedron","Width of the octahedron").Width=1.0
     obj.addProperty("App::PropertyLength","Height","Octahedron", "Height of the octahedron").Height=1.0
     obj.addProperty("Part::PropertyPartShape","Shape","Octahedron", "Shape of the octahedron")
     obj.Proxy = self

  def execute(self, fp):
     # Define six vetices for the shape
     v1 = FreeCAD.Vector(0,0,0)
     v2 = FreeCAD.Vector(fp.Length,0,0)
     v3 = FreeCAD.Vector(0,fp.Width,0)
     v4 = FreeCAD.Vector(fp.Length,fp.Width,0)
     v5 = FreeCAD.Vector(fp.Length/2,fp.Width/2,fp.Height/2)
     v6 = FreeCAD.Vector(fp.Length/2,fp.Width/2,-fp.Height/2)
     
     # Make the wires/faces
     f1 = self.make_face(v1,v2,v5)
     f2 = self.make_face(v2,v4,v5)
     f3 = self.make_face(v4,v3,v5)
     f4 = self.make_face(v3,v1,v5)
     f5 = self.make_face(v2,v1,v6)
     f6 = self.make_face(v4,v2,v6)
     f7 = self.make_face(v3,v4,v6)
     f8 = self.make_face(v1,v3,v6)
     shell=Part.makeShell([f1,f2,f3,f4,f5,f6,f7,f8])
     solid=Part.makeSolid(shell)
     fp.Shape = solid

  # helper mehod to create the faces
  def make_face(self,v1,v2,v3):
     wire = Part.makePolygon([v1,v2,v3,v1])
     face = Part.Face(wire)
     return face

class ViewProviderOctahedron:
  def __init__(self, obj):
     "Set this object to the proxy object of the actual view provider"
     obj.addProperty("App::PropertyColor","Color","Octahedron","Color of the octahedron").Color=(1.0,0.0,0.0)
     obj.Proxy = self

[... 163 lines cut ...]
Locale: English/United Kingdom (en_GB)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2020-04-28 19:45
tomkcook : Please address your issue according our [reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by first opening a topic on [our forum](https://forum.freecadweb.org/index.php) ("Python" subsection) so community can vet the issue and narrow down the causes. Cross-link the topic with this ticket. Thx

---

### Comment by tomkcook 2020-04-29 07:59
openBrain : I raised the problems described in [this thread](https://forum.freecadweb.org/viewtopic.php?f=22&t=45459&sid=bf41c0f5586091859db72e695dbc3d42) and got very little response.  If that thread reaches a resolution and you want to assign me a wiki login, I'm very happy to update the documentation myself.

---

### Comment by tomkcook 2020-04-30 10:52
Have also just verified the same result using the latest AppImage version downloaded from the FreeCAD website.

luzpaz, 2022-07-16:
Hi @onekk, would you be interested in attending to this issue?

onekk, 2022-07-22:
I could make something, but I haven't catch what should be done, make a working example?

Regards

Carlo D.

Roy-043, 2023-07-15:
The page has apparently been updated in the meantime. The mentioned examples now work.

closed 2023-07-15 by Roy-043