FreeCAD/FreeCAD #6054: Enable visual grouping of constraints

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4331

* **Reporter:** aster
* **Date submitted:** 4/30/2020
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Hello,

A few days ago we started a discussion on the forum about a visual improvement in how freeCAD shows constraints in the combo box. This feature may be enabled by checking a combo box in the above "Filter" menu.

https://forum.freecadweb.org/viewtopic.php?f=8&t=45753&p=392705#p392705

Here, for example, I should see only 4 constraints, ideally:
Constraint 1, 2, 3
Constraint 4, 5
Constraint 6, 7
Constraint 8

!(https://forum.freecadweb.org/download/file.php?id=111955&sid=c137169ea44d3253c447f80856291f73)

I created the FR with low priority and severity trivial, feel free to change

# FreeCAD Info

```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16131 (Git)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 3129ae4296e40ed20e7b3d460b86e6969acbe1c3
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Italian/Italy (it_IT)

```

# Other bug information

* **Priority:** low
* **Severity:** trivial
* **Category:** Feature
* **Updated:** 2/6/2021

adrianinsaval, 2023-04-21:
request has been dormant for 3 years and did not have community consensus to begin with, I would close this for now and if there are interested people they may reopen or better create a new issue following the new process. Thoughts?

@abdullahtahiriyo do you think this is a request worth keeping around?

abdullahtahiriyo, 2023-05-02:
Honestly I do not see a clear future for this, as it appears to contradict quite some other design decisions, which do not seem to be flawed. I think it makes sense to close it.

Thanks.

closed 2023-05-02 by adrianinsaval