FreeCAD/FreeCAD #6062: grouped buttons conflict in customized toolbar

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4342

* **Reporter:** fc_tofu
* **Date submitted:** 5/14/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Use toolbar customize funtion. I dupliated PartWB toolbars into DraftWB.
Before restarting FreeCAD, everything works ok.
But after restarting FreeCAD, grouped buttons dont display name and dont work when pressed, on both DraftWB and PartWB sides.

# Additional information

Forum:
[https://forum.freecadweb.org/viewtopic.php?f=34&t=42799&p=363857#p363857](https://forum.freecadweb.org/viewtopic.php?f=34&t=42799&p=363857#p363857)

# Steps to reproduce

1. Restart FreeCAD, and switch to DraftWB.
2. menu >Tools >Customize >Toolbars.
3. on right pane, add a toolbar named "Draft Custom" (for example).
4. on left pane, find "Part >Compound Tools" (this is a tool group).
5. add "Compound Tools" to "Draft Custom".
6. Close "Customize" and restart FreeCAD.
7. Obsever "Compound Tools" on DraftWB and PartWB toolbars.
!(https://i.imgur.com/v3z9Noi.jpg)

!(https://i.imgur.com/SNuYJRO.jpg)

!(https://i.imgur.com/c3x4O8g.jpg)

!(https://i.imgur.com/x7v24qZ.jpg)

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21049 (Git)
Build type: Release
Branch: master
Hash: 9c27f1078e5ec516fe882aac1a27f5c6c6174554
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Chinese/China (zh_CN)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

Roy-043, 2022-05-03:
Confirmed.

The OP has attached a user.cfg file to the forum thread:
https://forum.freecadweb.org/viewtopic.php?p=401193#p401193

Here is a smaller user.cfg file that demonstrates the problem:
[user-roy.zip](https://github.com/FreeCAD/FreeCAD/files/8614459/user-roy.zip)

```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.28765 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

Roy-043, 2024-12-15:
Issue still exists.

Jonezzzzz, 2025-01-05:
I looked into this issue and found the following:

The tools `Explode compound` and `Compound filter` are implemented in Python, while `Make compound` and the `GroupCommand` are implemented in C++.

The Python-based tools are only imported when the Part workbench is loaded (opened). This means that if you switch to the Draft workbench without first loading the Part workbench, these tools are unavailable.

The C++ based parts are available without loading the Part workbench and that’s why the tool is only partially loaded.

For example, if you create a custom toolbar and add a pure Python-based tool like the `Shaft Design Wizard` from the Part Design workbench to the Draft workbench, the tool won't appear unless the Part Design workbench has been loaded at least once.

However, once you load the Part Design workbench, the tool becomes available and works correctly in both Draft and Part Design workbenches.
(To reproduce this, you need to change the startup workbench.)

### Potential solutions
1.	 When a tool from workbench A is added to workbench B, workbench A could automatically load   when workbench B is opened.
**Problem**: This approach breaks down for tools which are implemented in both, Python and C++, if they are added both ways (e.g., A tools in B, and B tools in A), as both workbenches cannot be loaded simultaneously.

2.	Avoid tools that are implemented partly in C++ and partly in Python. Keeping the implementation in a single language could prevent inconsistencies in loading behavior.
**Problem**: Tools implemented in Python still need the owner(?) workbench to be loaded first.