FreeCAD/FreeCAD #6065: Pad or Pocket of ShapeString in Part Design causes BOPcheck to abort (OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4350

* **Reporter:** TedM
* **Date submitted:** 5/28/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** #ready-for-migration, 0.19, OCC Bug, OCC NotOnRoadmap, Part Design, shapestring, upstream

# Original report text

[Forum link](https://forum.freecadweb.org/viewtopic.php?f=3&t=46838)

After successfully making a pocket or pad of a ShapeString within a Part Design body, BOP Check fails with "Solid1 : Solid : BOPAlgo OperationAborted".

# Additional information

Also tested under release 0.18.4 and "Analyzes Geometry for Errors" with BOPCheck enabled simply causes FreeCAD to crash (program closes).

The examples use an open-source font from [Google](https://fonts.google.com/specimen/Roboto?subset=latin&sidebar.open&selection.family=Roboto) for portability. My original encounter was using a native Windows font.

Note: It has been suggested that this may be related to an upstream OCC bug: https://tracker.dev.opencascade.org/view.php?id=31522

Note: Tracking this ticket and upstream in [OCC bugs in FreeCAD](https://forum.freecadweb.org/viewtopic.php?t=20264) forum thread

# Steps to reproduce

1. Create new document.
2. Go to Part Design workbench and create a body.
3. Create a sketch of a fully constrained rectangle and pad it.
4. Go to Draft workbench and create a ShapeString, small enough to fit in the rectangle.
5. Go back to Part Design workbench.
6. Move the ShapeString into the body and position so it is sitting on the surface of the pad.
7. Using the shapeString, either make a pocket into (but not through) the solid or make a pad onto the surface of the solid.
8. Go to Part workbench.
9. Select the body and select "Analyzes Geometry for Errors".
10. Check "Run BOP Check" and, under "Individual BOP Checks", check "Self-Intersect".
11. Click "Run Check".
12.  Error result is "... Solid : BOPAlgo OperationAborted".

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21125 (Git)
Build type: Release
Branch: master
Hash: 40600a55c2fe71ff589be677f6e427ccc937d003
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 Home  1909 18363.836**
* **Platform:** Windows
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by TedM 2020-05-30 07:21
When reading through the forum thread, please ignore early replies disputing the problem. There was confusion from some of the posters where they were looking at a suggested workaround file instead of my original.

Further testing reported on the forum thread show these results:

* Lone ShapeString => No error
* ShapeString with Part/Extrude => Error
* ShapeString with PartDesign/Pad => Error
* ShapeString with PartDesign/Pocket => Error
* Any failing case above exported as STEP then reimported => No error
* ShapeString converted to sketch then Extrude/Pad/Pocket => No error
* Error only occurs if the text/font includes curves. Text without curves, for example "TTTTkkkk" in a sans-serif font => No error 

It has been suggested that this may be a duplicate of [https://tracker.dev.opencascade.org/view.php?id=31522](https://tracker.dev.opencascade.org/view.php?id=31522)

chrisb reports that the problem also occurs under MacOS:
OS: macOS High Sierra (10.13)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21125 (Git)
Build type: Release
Branch: master
Hash: 40600a55c2fe71ff589be677f6e427ccc937d003
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

---

### Comment by Kunda1 2020-05-30 15:20
Updated ticket to reflect it may be an upstream bug. Please notate ticket if in fact it is an I will track both this ticket and upstream via a forum thread that tracks OCC bugs in FreeCAD: https://forum.freecadweb.org/viewtopic.php?t=20264

---

### Comment by Kunda1 2020-12-31 12:20
Can someone bump the upstream ticket, please ? (and also confirm this exists on OCC 7.5.0?)
This ticket is being tracked from [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264&start=40) forum thread.

luzpaz, 2022-05-31:
Upstream ticket needs to be bumped. Any volunteers?

grubertechnic, 2023-07-29:
@luzpaz or @Roy-043

The BOPcheck is now successful, that shape have no issues, the operations works fine.

It was solved due FreeCAD, not OCCT see https://forum.freecad.org/viewtopic.php?p=404159#p404159

But in general fonts can be invalid for CAD, because they are not often precise enough for boolean operations and/or have intersecting and/or open wires. When i read this site https://forum.freecad.org/viewtopic.php?t=46838&start=20, maybe the font itself is the issue, at least i never read directly, that OCCT is per se the issue. It can be, that the issues are only symptoms of a fault font.

attached files to testing/playing
[6065.zip](https://github.com/FreeCAD/FreeCAD/files/12207601/6065.zip)

Upstream or not, it works now, imho a candidate to close.

grubertechnic, 2023-07-29:
forgotten the version

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33658 (Git)
Build type: Release
Branch: master
Hash: ba5f7e6d75dee09ec7076ad24895b04e56f21b63
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
```

Roy-043, 2023-07-31:
Thanks. Closing as resolved then.

closed 2023-07-31 by Roy-043

luzpaz, 2023-07-31:
OCC forum thread needs to be updated https://forum.freecad.org/viewtopic.php?t=20264