FreeCAD/FreeCAD #6067: CMake Install process needs revamping

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4352

* **Reporter:** hobbes1069
* **Date submitted:** 5/29/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** cmake

# Original report text

I've posted about this some time ago (i.e. 2011) either to the bug tracker or the forums but that's irrelevant now. It looks like there have been some changes in 0.19 prerelease that seem to move in the right direction but the work is incomplete. My current spec file for Fedora no longer installs files to the correct location and my old method of dumping almost everything into /usr/lib{,64}/freecad no long functions.

Having a fully compliant FHS install is long overdue and while I can't find the specific messages anymore, it was promised some time ago.

I am submitting this to the issue tracker in hopes of getting the attention it deserves, and part of that is offering my time to help fix it ONLY if there is sufficient interest from the developers. To be honest, I don't have time for this. I'm working on cutting back on my Fedora packaging duties due to $LIFE and $DAYJOB, but I really don't like the way things are right now and am willing to make the short term sacrifice, but again, only if it is taken seriously.

1. For the love of god, please use CMake's GNUInstallDirs module. This will take care of all *Nix systems and maybe others. 

2. Unless the libraries produced by FreeCAD are intended for TRULY shared (ie.e. public) consumption, they belong in a private subdir, i.e., /usr/lib{,64}/freecad on RPM (FHS compliant systems), but really all you have to do is append /freecad to CMAKE_INSTALL_LIBDIR if you use GNUInstallDirs, it takes care of the odd Debian tuple location for libraries. 

Certainly the libraries not prefixed with "lib" can't go into /usr/lib{,64}.

There are 93 instances of CMAKE_INSTALL_LIBDIR being used directly based on current master:

$ grep -ro CMAKE_INSTALL_LIBDIR | wc -l
93

My recommendation would be instead of sed'ing these all to "CMAKE_INSTALL_LINDIR/freecad" is to replace them with "FREECAD_INSTALL_LIBDIR" and default them to the correct location in one of the helper macros.

3. If you do this, CMake is smart enough to know if the RPATH needs to be set or not, negating the need for all the manual RPAH handling.

4. All no-arch data does belong in /usr/share, but it ALSO needs to be in a "freecad" subdirectory. Currently if I don't override it a lot of stuff is dumped directly into /usr/share, like some of the Mod python files.

It can be handled similarly to #2. Also, I haven't dug into it enough to know it's not being done correctly but stuff in /usr/share REALLY needs to be noarch. Meaning it's EXACTLY the same if its i686, x86_64, ppc64le, s390x, arm, etc. 

Now that I actually cut out the install portion of the build log and sorted it. It's not quite as bad as I thought, but my premis still stands. I'll attach the install portion so you can see where things are being installed when no override is being specified on the command line. 

# Additional information

Side topic: 
Maybe it matters for other platforms but I don't understand the overriding the output location of binaries and libraries in cMake/FreeCadMacros.cmake. By default they are created in the same directory they are configured in and CMake can track this just fine.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Platform:** ALL
* **Updated:** 3/21/2021

# Discussion from Mantis ticket

### Comment by hobbes1069 2020-05-29 13:44
Ok, after several build attempts yesterday trying to beat the package back into submission I got more than a little frustrated. With fresh eyes I found GNUInstallDirs buried in "cMake/FreeCAD_Helpers/ConfigureCMakeVariables.cmake". 

It's also in "src/XDGData/CMakeLists.txt" which I would say would be inappropriate to include two different places but my guess this is an attempt to override the PREFIX in the former location:

        set(CMAKE_INSTALL_PREFIX "/usr/lib${LIB_SUFFIX}/freecad" CACHE PATH "Installation root directory")

I believe this is only necessary because all of the Mod/ installs use the relative install path "Mod/<module>" which will only be installed relative to the PREFIX. 

Also, I misspoke in the original post, noarch data if it's executable code still belongs in /usr/lib, not /usr/share, which IS for noarch data, just not noarch code such as pure python.

Evaluating the install location portion of 

```
macro(ConfigureCMakeVariables)
    # ================================================================================
    # Output directories for install target
[... 71 lines cut ...]

There's still some non-script stuff being installed to /usr/lib{,64}/freecad/Mod I'll work on that when I have a little more time.

---

### Comment by vocx 2020-06-02 16:21
[quote=hobbes1069]Why are the translations (I assume that's what the qch and qhc files are) installing to /usr/doc? That doesn't make any sense.
[/quote]

These files are not translation files, they are compiled HTML files. They comprise the offline documentation that can be opened with FreeCAD, and need Qt assistant to be installed, I think.

---

### Comment by hobbes1069 2020-06-02 16:25
I figured that out later, I still question if they should get installed into /usr/doc, which is typically for plain text, markdown, rtf, files. Since they are compiled I would consider them to be a Qt resource file and should go somewhere in /usr/share/freecad.

---

### Comment by PrzemoF 2020-07-01 09:18
Default file locations on fedora linux.

---

### Comment by PrzemoF 2020-07-01 09:18
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=47159

luzpaz, 2023-02-26:
Is this ticket still relevant in lieu of all the CMake changes made recently ? (CC @berniev)

berniev, 2023-02-27:
No opinion.

closed 2024-12-30 by github-actions[bot]