FreeCAD/FreeCAD #6073: Tube does not show Fixed or Force Constraints

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4366

* **Reporter:** FreeCAD97990
* **Date submitted:** 6/8/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

I've noticed that Fixed Constraints and Force Constraints are not displayed when they are specifed on the end faces of a Tube.

Please see the attached file.

I've looked at a number of different cross sections, like a solid bar, and also tubes with square, hexagonal, and octagonal hollow profiles, and they all show Fixed Constraints and Force Constraints properly.

I was at first concerned that because the Constraints were not shown on the screen then the Constraints had not been applied to the Calculix model either. But when I went on to mesh and solve the Analysis, then there is deformation that shows that the Fixed and Force Constraints have actually been applied. They also appear in the Calculix inp file.

I put out a Forum post on this issue on Thu May 14, 2020 6:26 pm and bernd replied on Thu May 14, 2020 8:34 pm to confirm the bug and to say that it seemed to occur when the walls of the Tube are thin (less than around 10mm). He said we should make a entry on mantis and post a link to it here.

Now, I'm not too good at making links, so please refer to these posts if I haven't managed to put a link into this bug report.

# Steps to reproduce

Just load the attached file. No constraints are shown. I've just tested it with the latest 0.19 release, and the problem is still there.

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21514 (Git)
Build type: Release
Branch: master
Hash: 89ddab33f692a9bd739171689f9ab0eb58d64b35
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United Kingdom (en_GB)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

luzpaz, 2023-02-18:
@FreeCAD-Tools do you mind testing this to see if it's still a valid bug ?

FreeCAD-Tools, 2023-03-06:
Nope. Needs a specialist in FEM here.

FEA-eng, 2023-12-04:
I wanted to test this in 0.22 but I just get this error when trying to open the file:

`Invalid project file: ios_base::failbit set: iostream stream error`

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

marioalexis84, 2023-12-22:
I confirm the bug and found the problem.
Working on it.

FEA-eng, 2023-12-24:
> I confirm the bug and found the problem. Working on it.

A similar issue has been reported recently on the forum: https://forum.freecad.org/viewtopic.php?t=83613

Perhaps, the cause is the same.

marioalexis84, 2023-12-24:
> A similar issue has been reported recently 
Yes, is the same.

commit 27a0fb1e (2024-01-16): Fem: Fix constraint arrows - fixes #6073
closed 2024-01-16 by chennes