FreeCAD/FreeCAD #6074: Duplicated window object when using expression with wall width

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4370

* **Reporter:** vocx
* **Date submitted:** 6/11/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

An Arch Window can be embedded into an Arch Wall. Normally in the tree view this relationship is shown as the Window being inside the Wall.

If using expressions to define some properties of the Window, the Window is duplicated in the tree view, although only one Window object actually exists.

The bug may be in the base system actually, that is, in the code that handles the viewproviders or the tree view, and not in the Arch Wall or Arch Window object.

Forum thread: [Bug: duplicated window object when using expression with wall width](https://forum.freecadweb.org/viewtopic.php?f=23&t=47169)

# Steps to reproduce

1. Click on Arch Wall, and create a wall 3 m, 3 m, and 0.2 m. The wall should be big enough to hold a standard window.
2. Click on Arch Window, Fixed, and place it on the wall.
3. In the tree view click on the Window, go to the property editor, choose "Hole Depth", and click on the rightmost "f(x)" symbol to set an expression (expressions).
4. The expression is just the Wall's width, so enter Wall.Width in the expression field, and click OK.
5. The Window will show the blue mark indicating the need for recomputing, so recompute, F5 (Std_Refresh).
6. The Wall doesn't show a blue mark. Nevertheless, right click on it on the tree view, and then choose "Recompute object".
7. The Window will appear a second time in the tree view. There will be two windows, and both will have the blue checkmark.

# FreeCAD Info

```
OS: Ubuntu 18.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21562 (Git)
Build type: Release
Branch: master
Hash: 11a2b948ea55b19efe7515f730035027a7822e06
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 18.04**
* **Platform:** Linux
* **Updated:** 2/6/2021

luzpaz, 2022-05-03:
Per @yorikvanhavre in https://forum.freecadweb.org/viewtopic.php?p=405231#p405231
> Objects can appear many times in the tree view, that's normal.
> But the recomputation indeed has something weird, I'll have a better look. In any case, the expression engine works with the dependency tree, so if there was a circular dependency, you'd see it in the dag view, so it's probably not the case

@Roy-043 then mentions that he can't reproduce in v0.19.24219

Roy-043, 2022-05-04:
Still can't reproduce. IMO this issue can be closed.
```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.28765 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

yorikvanhavre, 2022-05-20:
I cannot reproduce either, and tried all kinds of tricks to make it appear, nothing seems unusual. Let's close it I guess...

closed 2022-05-20 by yorikvanhavre