FreeCAD/FreeCAD #6075: Crash when create FEM pipeline

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4373

* **Reporter:** david9991
* **Date submitted:** 6/15/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** feedback
* **Tags:** #pending-forum

# Original report text

Crash when create FEM pipeline

# Additional information

See backtrace in attached log3.txt

# Steps to reproduce

1. Select a FEM result
2. Click the 'Create post processing pipeline from a result object' on toolbar.
3. Crash.

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21514 (Git)
Build type: Release
Branch: master
Hash: 89ddab33f692a9bd739171689f9ab0eb58d64b35
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by berndhahnebach 2020-06-22 13:36
Would you try if it works for you with the FemCalculixCantilever3D example from StartWB?

This one works great for me on Linux and Windows. I can not test on OSX.

bernd

BTW: It would be better to post on the forum first before creating a bugreport. See HUGE letters Attention on top of every FreeCAD mantis page.

---

### Comment by Kunda1 2020-06-26 17:05
david9991 please follow up + next time please post to the forum before opening a ticket. 

---

### Comment by Kunda1 2020-10-10 21:11
david9991 ping 
please use latest v0.19 update

---

### Comment by Kunda1 2020-10-14 02:30
bernd who on the FEM subforum runs macOS and can test these FEM macOS-related tickets ?

---

### Comment by Kunda1 2020-10-14 02:30
berndhahnebach ^

---

### Comment by Kunda1 2020-10-15 14:08
Discussion thread: https://forum.freecadweb.org/viewtopic.php?f=18&t=51186

---

### Comment by yorik 2021-12-15 09:46
Putting the backtrace in OP in an attached txt file to workaround export issues

---

### Comment by Kunda1 2022-01-24 20:08
bumped forum thread again

---

### Comment by Kunda1 2022-01-24 20:10
chrisb, could you please try to recreate on macOS ?

luzpaz, 2022-03-17:
@PLChris we need a macOS user to try to reproduce issue? Are you available?

donovaly, 2022-03-26:
@chennes , can you as MacOS user reproduce this issue?

donovaly, 2022-03-26:
I fixed yesterday some Pipeline bugs but could not encounter any crashes.

So unless anybody can reproduce this, we can close this bug, that is also old and the Pipeline code changed meanwhile.

luzpaz, 2022-03-26:
CC @PLChris

donovaly, 2022-03-28:
Nobody can reproduce this and the bug is 2 years old. The code changed meanwhile.

Closing.
If anybody can reproduce, please reopen

closed 2022-03-28 by donovaly