FreeCAD/FreeCAD #6077: Crash after changing a pocket-size

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4378

* **Reporter:** cappu
* **Date submitted:** 6/18/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** #pending-forum, 0.19, crash, Part Design

# Original report text

Crash after changing a size-parameter in a sketch and close sketch (cpu gets 100% load and memory runs full)
I am not sure if ths issue is found before as I do not know how to name it.
But on my PC it is very simple to reproduce.

# Additional information

I use appimage file.
If size of hight-parameter is less than 33 mm, no problems occur. 33mm and bigger seems to be a problem which is not catch in code.
I created this 3d-model last week and updated appimage-version 1 time. With current version of FreeCad and at least one version before the issue occurs on my PC.
Hope you can reproduce it on your PC.

# Steps to reproduce

1) open RegalBuero (copy).FCStd
2) change WS to part-design
3) open last sketch043:  RegalBuero/P2_KeinteilBoxen../B8_BodenHinten001/Seitenloecher/Sketch043

4) In this sketch043 change the very right-side hight-parameter from 32,8 to 33mm
5) close the sketch
6) look on cpu and RAM.  On my PC CPU become 100% load (after aprox 5sec) and RAM starts slowly overflow (approx 30sec left until complete overflow)

# FreeCAD Info

```
OS: Ubuntu 20.04 LTS (XFCE/xubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21654 (Git) AppImage
Build type: Release
Branch: master
Hash: ddf0cf3136a6c5a459abae4bf1f9c714f6f45c86
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **OS: Xubuntu  20.04**
* **Platform:** linux
* **Updated:** 3/3/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2020-06-26 17:09
cappu where is the forum thread (as per our bugtracker guidelines)?
Please open one next time before opening a ticket. Please open a thread now and link back to this ticket. Also add the forum thread to this ticket so we can track it. 

---

### Comment by Kunda1 2020-06-26 17:10
Also there is no file (RegalBuero (copy).FCStd) attached to this ticket to test

---

### Comment by cappu 2020-06-27 10:58
Hello, 
Thanks for taking care.
Please find attached file (hope it works now)
No thread at the moment because in description it is description it is described if clearly a bug, it can be directly added here. But I will create a forum thread...
I tried today with latest version, and crash still happens on my PC

OS: Ubuntu 20.04 LTS (XFCE/xubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21764 (Git) AppImage
Build type: Release
Branch: master
Hash: ccea55a65512d37081d708d6f504d0c4d182a9e0
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Best regars

---

### Comment by cappu 2020-06-27 11:06
Hm, looks like I do something wrong at adding the file. Filesize 1,6MB. No error. Green bar under file.

But when I go over the attached file is says:
html> <head><title>413 Request Entity Too Large</title></head> <body bgcolor="white"> <center><h1>413 Request Entity Too Large</h1>

---

### Comment by cappu 2020-06-27 11:07
I already removed most parts of the model. Is there a different way to add the file?

---

### Comment by cappu 2020-06-27 11:22
Tried to create a forum thread. But here an error comes when trying to add the file. 
(see screenshot)
But without file, it does not make sense to talk about (Description: On Increasing a pocket size, computer crashes???  ). 
Is there a possibility to send you the model?

---

### Comment by cappu 2020-06-27 12:50
Datei siehe

https://www.magentacloud.de/lnk/ahAhDTWN

---

### Comment by cappu 2020-06-27 13:21
Forum Link:   https://forum.freecadweb.org/viewtopic.php?f=19&t=48101
(hope it is the correct forum. I have not found one to verify bugs)

---

### Comment by Kunda1 2020-07-02 18:07
Related partial fix: c:FreeCAD:31b1ad874fb4750:

---

### Comment by Kunda1 2021-03-03 11:51
Trying to add attachment

luzpaz, 2022-04-01:
Bumped [forum thread](https://forum.freecadweb.org/viewtopic.php?p=585315#p585315)

luzpaz, 2022-09-08:
No response from OP or file available to test (3rd party hosted example file is unavailable). Can't reproduce.
Closing ticket. If OP resurfaces, please re-open ticket.

closed 2022-09-08 by luzpaz