FreeCAD/FreeCAD #6078: DraftWB, error in Array position when "Link Array" checked off

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4382

* **Reporter:** fc_tofu
* **Date submitted:** 6/22/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 0.19, array

# Original report text

DraftWB, Array operator, when "Link Array" option checked off, error appear on viewport.

# Additional information

Forum:
[https://forum.freecadweb.org/viewtopic.php?f=3&t=47471](https://forum.freecadweb.org/viewtopic.php?f=3&t=47471)

# Steps to reproduce

1. Startup FreeCAD, switch to DraftWB
2. Draw a circle
3. invoke "Array" operator
4. Task panel of "Array", check off "Link Array", OK
5. Observe on viewport, strange points appear and placement of array instance is wrong.

!(https://forum.freecadweb.org/download/file.php?id=117476)

!(https://forum.freecadweb.org/download/file.php?id=117477)

!(https://forum.freecadweb.org/download/file.php?id=117478)

Strange points seems center of circles, the only vertex of circle. So, the issue means vertices and edges are separated after non-link Array.

In case of "Link array" checked on, the result looks right in viewport. Then after Downgrade, same result (as Reproduce step 5) happens.

Recompute, restart FC, restart PC, reset "user.cfg" donn't help in both cases.
This bug doesnot happen to FC0.18.4/Win10x64.

From discussion, this bug seems related to OCCT 7.4.0, not to 7.3.0.

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21329 (Git)
Build type: Release
Branch: master
Hash: 2330eef823b32ac412d839031cc174353a76b013
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Chinese/China (zh_CN)
```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

luzpaz, 2022-02-24:
Draft 'Array' related #6030

luzpaz, 2022-02-24:
Closing as it is not reproducible per the forum thread

closed 2022-02-24 by luzpaz