FreeCAD/FreeCAD #6081: Start paraview with selected result from within FC

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4388

* **Reporter:** HoWil
* **Date submitted:** 6/29/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** confirmed
* **Tags:** Elmer, FEM, Paraview, postprocessing, VTK

# Original report text

I would like to have the possibility to use paraview from (within) FC. Paraview is 'the' tool for visualizing FEM-Results so why shouldn't we use it in combination with FC-FEM.
Paraview is already heavily in use for Cfd-OF/Openfoam within the FC-community.
The results created by Elmer when used in FC-FEM are stored in .vtk format. By storing e.g. a paraview-state-file pvsm with the same filename next to the .vtk file one could automate nice post processing. In future such paraview-state-information or equivalent python-code could be stored and edited within FreeCAD in a document-object. 

# Additional information

https://forum.freecadweb.org/viewtopic.php?f=18&t=43278#p367842

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2022-12-03:
Per HoWil on the forum:
> I have created a maro-file which can be used to:
> 1. automatically open the .vtu file created by Elmer solver in Paraview (until now supporting only "Beside .FCStd file" setting in FEM-Preferences)
> 2.  the configuration section in the macro offers support for the use of a .py or .pvsm state file which has to be loacted in the same folder as the .vtu file.
> 3. several ElmerSolver objects in a .FCStd file are supported

> Simply select the ElmerSolver object and run the macro.

>Limitations

> * everytime one runs the macro a new instance of Paraview will be opened. To circumvent that one can open the .vtu file in Paraview with or without using statefile and reload the 'Reload files' every time the results are regenerated.
> *  It was not tested on Window

```python
# -*- coding: utf-8 -*-
## Paraview postroccesing started from within FreeCAD
# Author: HoWil
# License: LGPL v 2.1
# Version: 20200703

# =============================================================================
# Description
# =============================================================================
# First, one hast to select a ElmerSolver element in the Tree-View in FreeCAD.
# Second, check the configuration below and run the macor to open the 
# 'case_t0001.vtu' generated by ElmerSolver with the specified existing 
# state-file in Paraview.

# =============================================================================
# CONFIGURATION
# =============================================================================

# Tested on Ubuntu 19.10

optional_path_to_paraview = ''  # if empty use the installed paraview;
# e.g. '/home/user/Downloads/ParaView-5.2.0-Qt4-OpenGL2-MPI-Linux-64bit/bin/'
#optional_state_file = 'statefile.pvsm'  # has to located next to the .vtu
#optional_state_file = 'statefile.py'  # has to located next to the .vtu
# optional_state_file = ''  # please use an empty string if not needed

# Windows not yet tested

# =============================================================================
# START MACRO
# =============================================================================

import FreeCADGui
import FreeCAD
from PySide import QtCore, QtGui
from platform import system
import os
import Fem

# Get the selected objects
selection = FreeCADGui.Selection.getSelectionEx()

if len(selection) == 1: # test if only one object is selected

    sel = selection[0] #
    print('\nSelected Object: ', sel.FullName)

    if sel.TypeName == 'Fem::FemSolverObjectPython' and hasattr(sel.Object , 'ElmerOutput'):  # test if the selected object is a solver object and has an 'ElmerOutput' attibute
        solver = sel
        print(solver.Object.FullName)

        objectName = solver.ObjectName

        filename, file_extension = os.path.splitext(sel.Object.Document.FileName)
        head, tail = os.path.split(filename)

        resultfile_dir = head + '/' + tail + '/' + objectName

        print(os.getcwd())
        os.chdir(resultfile_dir)
        print(os.getcwd())

        file_exists = False
        if 'optional_state_file' in locals() and optional_state_file is not "":
            state_file_param = resultfile_dir + '/' + optional_state_file
            if not os.path.isfile(state_file_param):
                FreeCAD.Console.PrintError('Error!! State file could not be found.')
                QtGui.QMessageBox.critical(None, 'Error', 'Error!! State file could not be found. Please doublecheck your configuration in the macro file or report the error in the FreeCAD forum.', QtGui.QMessageBox.Abort)
            else:
                file_exists = True
                paraview_command = optional_path_to_paraview + \
                    'paraview --state="' + state_file_param + '"'
        else:
            resultfilename_vtk = resultfile_dir + "/case_t0001.vtu"
            if not os.path.isfile(resultfilename_vtk):
                FreeCAD.Console.PrintError('Error!! Result file could not be found.')
                QtGui.QMessageBox.critical(None, 'Error', 'Error!! Result file could not be found. Please doublecheck your configuration in the macro file or report the error in the FreeCAD forum.', QtGui.QMessageBox.Abort)
            else:
                file_exists = True
                paraview_command = optional_path_to_paraview + \
                    'paraview --data="' + resultfilename_vtk + '"'
                
        if file_exists:
            try:
                process = QtCore.QProcess()
                process.startDetached(paraview_command)
                FreeCAD.Console.PrintMessage('Running ' + paraview_command + '\n')
                FreeCAD.Console.PrintMessage('Finished '+ paraview_command + '\n')
            except:
                FreeCAD.Console.PrintError('Error!!, sorry..')
                QtGui.QMessageBox.critical(None, 'Error', 'Error!! Sorry. Please report the error in the FreeCAD forum', QtGui.QMessageBox.Abort)

    else:
       print('Please select a single ElmerSolver object')
       QtGui.QMessageBox.information(None, 'Information', 'Please select a single ElmerSolver object', QtGui.QMessageBox.Abort)

else:
   print('Please select a single ElmerSolver object')
   QtGui.QMessageBox.information(None, 'Information', 'Please select a single ElmerSolver object', QtGui.QMessageBox.Abort)
```
[statefile.zip](https://github.com/FreeCAD/FreeCAD/files/10146977/statefile.zip)
[Tutorial_nonGUI_6 - Electrostatics - StatElecSolve, ElectricForce - m-based_fromBody.zip](https://github.com/FreeCAD/FreeCAD/files/10146979/Tutorial_nonGUI_6.-.Electrostatics.-.StatElecSolve.ElectricForce.-.m-based_fromBody.zip)
[Peek 2020-07-05 11-30.webm](https://user-images.githubusercontent.com/4140247/205453243-8dd3257a-d98e-4994-8d6a-766b5e0573db.webm)

marioalexis84, 2026-03-05:
@FEA-eng is this necessary? Pipelines can be exported as VTK files and opened in ParaView.
I think this should be a macro; it could even include adding an item to the context menu or the utilities menu bar.
What we definitely need to add is the ability to export the results with time steps as .pvd files.