FreeCAD/FreeCAD #6089: scientific notation for displaying numbers

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4402

* **Reporter:** HoWil
* **Date submitted:** 7/8/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** display, formula, NumberFormat, Precision, Solver(FEM), spinbox

# Original report text

I ask for using the scientific notation for displaying numbers where it is necessary.
The reason is the following.
I often use FEM and among others have to edit solver settings in the range of 1e-5 to 1e-9. All of these settings are simply shown as 0.0.

I know that it is already possible to enter number in scientific notation. I ask for some additional switch in the preferences next to 'number of decimals called something like 'use scientific notation to display numbers where necessary?'

The formula editor can help here. One can insert 1e-9 and the following is shown in the Value-column in the Data-Tab: 0,00 (1e-6). But I have to use the formula editor first which means I have to add all necessary (pre-filled) values into the formula editor to see/recognize it in the Value-column.
Below you see an example where the entry for Linear Tolerance is 0,00000001 and is shown as "0,0" where 1e-8 would be nice. When I would like to change it without formula editor I would have to set the number of decimals to 8 making other dialogues hard to use because of many unnecessary zeros.
Under Steady State Tolerance you can see the value "0,00 with (1e-9)" using the formula editor. 

# Additional information

Please see also :
https://forum.freecadweb.org/viewtopic.php?f=8&t=48229

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2023-02-26:
@donovaly is this on your FEM radar as well ?

donovaly, 2023-02-26:
I raised this issue already last year when working on FEM. The outcome is that the only way to use scientific notation in property editor is to use an expression. I described this in the Wiki already. Also, for the Elmer solvers I already implemented that an expression is used for settings like the "Linear Tolerance".

For technical reasons, the direct input of scientific notation to a property field cannot be implemented as @wwmayer wrote me. Unfortunately I cannot find right now the corresponding forum thread.

donovaly, 2023-02-26:
I found now the forum thread: https://forum.freecad.org/viewtopic.php?t=70646