FreeCAD/FreeCAD #6090: import PRC file or 3D PDF into FreeCAD

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4403

* **Reporter:** ferdymercury
* **Date submitted:** 7/8/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** conversion, file format, import, STEP

# Original report text

This is a follow up the suggestion I made [in the forum](https://forum.freecadweb.org/viewtopic.php?p=414775&sid=95bb42cdf58e965c1ea9cc0c9faee09d#p414775).

PRC files are commonly used by several vendors to embed 3D models in a PDF file. Unfortunately, these files can only be opened with Adobe PDF reader. For example, Linux does not have a native Adobe viewer, which means it is impossible to view correctly this type of documents. One needs some workarounds (Wine, Virtualbox, or old Adobe versions for Linux).

I thought that in general, not only for Linux users, it would be nice if FreeCAD would be able to import PRC files in the same way it imports STEP files. (Also including nested parts in the tree model).

It is easy to extract the embedded PRC file from the 3D PDF, by just searching for 'stream.PRC' and 'endstream' bytes in a hex editor, so in principle this could be done automatically also by FreeCAD. Everything inbetween is the PRC binary file that can be extracted, and for example included in your own PDF with pdflatex [media9](https://osl.ugr.es/CTAN/macros/latex/contrib/media9/doc/media9.pdf) package.

Thinking about a 'far future', if FreeCAD is able to open and import it, this would allow in the future that Linux PDF viewers display inline also 3D models, by embedding a small FreeCAD plugin directly, instead of having to open it in external FreeCAD window. In the same way than Adobe PDF reader has a Flash Player plugin to do that.

To implement this feature, one might reuse some parts of this code: https://github.com/XenonofArcticus/libPRC

The PRC file format is defined [here](https://web.archive.org/web/20091123055411/http://pdf.editme.com/files/PDFE/SC2N570-PRC-WD.pdf). 

Other posts related to generating PRC files from a FreeCAD model:
https://forum.freecadweb.org/viewtopic.php?t=20224&start=10
https://forum.freecadweb.org/viewtopic.php?f=8&t=18497#p144546
(This is the opposite direction than the feature requested here)

# Steps to reproduce

*  Download https://tetra4d.com/wp-content/uploads/2015/04/3D-PDF-Sample-Outlet.pdf
*  Open it with Bless hex editor or equivalent
*  Search for PRC keyword (search option Text, not Hex).
*  Select and delete in the hex editor from beginning of document until & including line above, containing keyword 'stream', so that first three bytes of the resulting file are 'PRC'.
*  Search for 'endstream' keyword. Delete from (including) this line until the end of the document.
*  Save result as .prc
*  In a Latex document, you can include it with
```
\usepackage{media9}
...
\includemedia[
  %label=dice, % name for referencing 3D object
  width=\linewidth,
  height=0.495\linewidth,
  3Dtoolbar,
  3Dmenu
  %activate=pagevisible,
  %3Dcoo=13.514159202575684 -210.37330627441406 265.50933837890625,
  %3Dc2c=-0.6669085025787354 0.39916229248046875 0.6292078495025635,
  %3Droo=9603.325257430366,
  %3Droll=-110.37121928895962,
  %3Dortho=0.001394091872498393,
  %3Dlights=Cube
  ]{}{result.prc}
```

*  Compile with pdflatex, and open it with Adobe PDF Reader

(Side note: if a 3D PDF file has several 3D models embedded, this has to be done in a for loop. Or the user has to select in the import dialog one among those)
 
%
%

# FreeCAD Info

```
OS: Linux Mint 19.3 (MATE/mate)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21775 (Git) AppImage
Build type: Release
Branch: master
Hash: 1f741aa511e898849e46ed14515fce416fe32acb
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by ferdymercury 2020-07-08 14:13
The following link is more updated than the one from libPRC I posted before.
https://github.com/vectorgraphics/asymptote/tree/50fb809fdaf2383385f09884174262964a046156/prc

One could expand the code in asymptote repository and write the functions in the opposite direction (Deserialize, Read) instead of (Serialize, Write). So that Asymptote library could be used as an 'intermediary'. Then, one would 'only' need to convert Asymptote structure to FreeCAD structure.

See also:
https://forum.freecadweb.org/viewtopic.php?t=156

---

### Comment by wmayer 2020-12-10 16:49
https://forum.freecadweb.org/viewtopic.php?f=8&t=53116

---

### Comment by ferdymercury 2020-12-10 16:54
See also: https://github.com/vectorgraphics/asymptote/issues/167

ferdymercury, 2022-02-25:
See also https://github.com/XenonofArcticus/libPRC/issues/2#issuecomment-1027164300

chiragsingh1711, 2025-03-21:
Hello @maxmosk ,

I'm Chirag Singh, and I'm interested in contributing to this project. I have extensive experience with 3D software like Blender, Fusion360, and FreeCAD, specializing in automation scripting. While I haven't worked directly with 3D PDFs, I'm eager to apply my skills and learn new technologies for this project.

Could we discuss how my experience aligns with your needs and how I can best contribute?

Best regards,
Chirag Singh

maxmosk, 2025-03-23:
@chiragsingh1711 maybe you mean @maxwxyz

chiragsingh1711, 2025-03-23:
> [@chiragsingh1711](https://github.com/chiragsingh1711) maybe you mean [@maxwxyz](https://github.com/maxwxyz)

Oh yess ! my bad

luzpaz, 2025-08-25:
Looks like:
* https://github.com/FreeCAD/FreeCAD/pull/23335 

impacts this ticket. What would be left is libPRC.

@ferdymercury, on a cursory inspection it doesn't look like libPRC is active or has active forks. Is 3D PDF enough ITC to satisfy needs going forward ?

ferdymercury, 2025-08-25:
> it doesn't look like libPRC is active or has active forks.

Well... it seems that the best way to go would be to work instead on https://github.com/vectorgraphics/asymptote/
which does get maintenance / active development.

A new PRCtool could be created there that instead of exporting the PRC, it parses the PRC file and exports it as eg V3D file which can be directly imported into FreeCAD.

See https://github.com/XenonofArcticus/libPRC/issues/2#issuecomment-1027164300 where the PRC file is printed into a text-terminal console. So that program should be modified to create the new 3D structures rather than printing via cout.