FreeCAD/FreeCAD #6096: Allow Selection of Print Area and Scale

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4413

* **Reporter:** wandererfan
* **Date submitted:** 8/2/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Allow user to select part of display for printing. Allow user tho specify scale of printed output.

[Forum discussion.](https://forum.freecadweb.org/viewtopic.php?f=3&t=49094#p420505)

# FreeCAD Info

```
OS: Linux Mint 19 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: e727cd37548d7f38ea120670d2a28aba871a4aa1
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Canada (en_CA)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by clifcox 2020-12-17 17:58
Perhaps we can reuse some code from LibreCAD for this?
https://forum.freecadweb.org/viewtopic.php?style=4&f=3&t=49094&start=20#p458468

luzpaz, 2023-10-18:
Currently using librecad is a workaround. See clifcox's link above.

WandererFan, 2024-05-18:
We don't draw areas, we draw objects.  Scale is controlled by properties on the Page and on individual views. 

There is nothing to do here.

closed 2024-05-18 by WandererFan