FreeCAD/FreeCAD #6097: Toolbar-only command must be visible for custom shortcut to work

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4415

* **Reporter:** Roy_043
* **Date submitted:** 8/3/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

For some reason a toolbar-only command (= a command that is not included in any menu) must be visible for the assigned custom shortcut to work. This seems like a strange limitation. One of the reasons for assigning shortcuts can be that you do not want to display the toolbar.

# Additional information

Bug topic:
https://forum.freecadweb.org/viewtopic.php?f=3&t=45896

openBrain's analysis:
https://forum.freecadweb.org/viewtopic.php?f=3&t=48949#p419162

# Steps to reproduce

1. Invoke the Std_DlgCustomize command.
2. Assign a shortcut to the Std_Part command. The command can be found in the Structure category.
3. Test the shortcut with the Structure toolbar visible.
4. Result: OK.
5. Disable the Structure toolbar.
6. Re-test the shortcut.
7. Result: the shortcut does not work anymore.

# FreeCAD Info

```
OS: Windows 8.1 (6.3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22111 (Git)
Build type: Release
Branch: master
Hash: cb2099aa6bb287a8d7843eb70684cce79bdef26b
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Dutch/Netherlands (nl_NL)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

luzpaz, 2022-03-13:
@realthunder could you take a look at this (in lieu of your work on #6506)

0penBrain, 2022-03-13:
Just had another look but can't find anything significant (there are interesting things in Qt source code, but they can't apply here). Especially I can't find any significant difference between "New Part" and "New Body" that would explain the difference in behavior -- yes, I know one is added into a menu and not the other, I talk about technical state differences :smile: --. I'm afraid we are dealing with some Qt things. It would probably need a FreeCAD compiled with a debug version of Qt to investigate deeper... :neutral_face:

realthunder, 2022-03-14:
The differences in menu bar and tool bar is most likely a Qt bug. FreeCAD hides irrelevant menus in the menu bar when switching workbench, and does the same for tool bars, but Qt treats it differently for actions in hidden toolbar and menus. The above references commit fix the problem. See the commit comments for more details.

maxwxyz, 2024-04-21:
any news on this? Issue is still present in main and holds back an option for a toolbar cleanup.

wwmayer, 2024-04-22:
AFAICS this issue is fixed. Just follow the procedure above and you will see that the Part creation works now.

closed 2024-04-22 by maxwxyz