FreeCAD/FreeCAD #6098: Additive/Subtractive Pipe while in wireframe view leaves multiple features visible

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4417

* **Reporter:** TedM
* **Date submitted:** 8/5/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Creating an AdditivePipe or SubtractivePipe while in wireframe draw style results in both the Pipe feature and the previous tip feature being visible (in model view, both are shown visible and they overlap in the 3D view). (I've also seen this happen when working in No-shading draw style, but it doesn't always happen.)

In addition, at this point, toggling visibility of these features is incorrect. Toggling the new tip (the Pipe) has no effect on the visibility of the previous tip. Toggling the previous tip the first time removes it from the 3D view but it remains visible in the model view. From then on, toggling the previous tip works correctly.

This has been reproduced in other 0.19 versions but cannot be reproduced in 0.18.4.

# Steps to reproduce

1.  Create new document.
2.  Go to Part Design WB.
3.  Create new body.
4.  Create new sketch on XY plane.
5.  Place a 20mm x 20mm square centered on origin and close sketch.
6.  Pad it 20mm.
7.  Create a second sketch on XY plane.
8.  Place a 10mm x 10mm square centered on the origin and close sketch.
9.  Change attachment to Z=21mm.
10.  Create a third sketch, this time on the XZ plane.
11.  Create a vertical line, bottom end on Y axis and 10mm above the origin and length 21mm and close sketch.
12.  At this point, you have a 20x20x20mm cube with a 10x10mm square sketch floating 1mm above and a line sketch coming up through the top of the cube and ending in the center of the square. This is what is in the attached file.
13.  In the model view, make sure the pad and both sketches are visible.
14.  (Optional, but probably required) Switch draw type to wireframe.
15.  Select the square (second) sketch and activate SubtractivePipe.
16.  In Pipe Parameters, click object, select the vertical line in the 3D view and click OK.
17.  At this point, you should observe that BOTH the Pad and SubtractivePipe are visible in the model view.
18.  Change the draw style to flat lines to confirm that the pipe is hidden in the pad, except for the square on the surface.
19.  In the model view, select the Pad and toggle visibility, either from the menu or space bar. The 3D view now shows the cavity (SubtractivePipe) in the face of the cube (Pad) but the model view still shows both visible.
20.  Again, toggle visibility and now the SubtractivePipe goes not visible in both views.

The same procedure may be done with AdditivePipe instead, except, of course, it's harder to detect the fault in the 3D view.

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22130 (Git)
Build type: Release
Branch: master
Hash: dc65b055e5143a7d7349520da466d4d722df57e8
Python version: 3.8.5
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by TedM 2020-08-05 07:06
Forum discussion here: [https://forum.freecadweb.org/viewtopic.php?f=3&t=49021](https://forum.freecadweb.org/viewtopic.php?f=3&t=49021)

luzpaz, 2023-11-11:
@bgbsww do you have a moment to check this ticket out ?

bgbsww, 2023-11-11:
@luzpaz, yes, I'll look.

bgbsww, 2023-11-11:
Note:  Step 11 should be 11mm instead of 21mm.  Step 14 was not required.  Step 16 should be Path Object as there are two buttons.   That's all irrelevant if you like the PR fix proposed.

closed 2023-11-16 by adrianinsaval

bgbsww, 2024-01-22:
Looking at this again, both Pipe and Loft show this symptom, while Pocket and Revolution/Groove do not.

Select the second sketch, and then any of the Additive or Subtractive buttons.  As soon as the Task dialog comes up, click back to the model.  Look to see what is enabled:

Pad:  The original Pad is off, the new Pad001 is on.  If you press the escape key, you exit the dialog.
Revolution:  original Pad is off, new Revolution is on.   Escape works.
Loft:  original Pad is ON, new AdditiveLoft is OFF.  Escape key does nothing, must return to tasks and hit cancel.
Pipe:  original Pad is ON,  new AdditivePipe is on.   Escape key does nothing
Helix:  original Pad is ON, new AdditiveHelix is on.  Escape key does nothing.
Primitives:  original Pad is ON,  new x is on.  Escape key does nothing.
Pocket:  Pad off, new Pocket on  Escape works
Hole:  Pad off, new  Hole on  Escape DOESN'T work.
Groove:  Pad off, new Groove on, Escape works
SubLoft:  Pad ON, new SubtractiveLoft OFF,  Escape does nothing.
SubPipe:  Pad ON, new SubtractivePipe on,  Escape does nothing.
SubHelix:  Pad ON, new SubtractiveHelix on,   Escape does nothing.
SubPrimitives:  Pad ON, new x on, Escape does nothing.

So,  the issue appears to occur right when the dialogs are first triggered for Loft/Pipe/Helix/Primitive; the Pad should get disabled at that moment.  And there is an adidtional issue with the Escape key not being detected for the same, plus hole.

Onward to look at the code ...

PR #11357 (2023-11-16): Refresh body after pipe created
Fix #6098