FreeCAD/FreeCAD #6099: osx line width not getting wider

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4418

* **Reporter:** brickmaker
* **Date submitted:** 8/15/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

I have currently tried to draw a simple line in a couple different workbenches including Arch and Draft. My issue I am having is when I draw a line and I want to change the line width it can't get wider than 7px? I ran the same test on a Ubuntu computer I have and it was able to draw a line with a width bigger than 7px. Thank you for your help.

[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=49435)

# Additional information

***Report View***
18:54:33  /Applications/FreeCAD.app/Contents/Resources/lib/python3.8/site-packages/mpmath/ctx_mp_python.py:892: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if other is 0:
18:54:33  /Applications/FreeCAD.app/Contents/Resources/lib/python3.8/site-packages/mpmath/ctx_mp_python.py:986: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if other is 0:
18:55:59  ----------------
18:55:59  GuiCommand: Line
18:55:59  Pick first point
18:56:01  Pick next point
22:43:10  ----------------
22:43:10  GuiCommand: Toggle grid
22:43:22  ----------------
22:43:22  GuiCommand: Line
22:43:22  Pick first point
22:43:35  Pick next point

***Python Console***
Python 3.8.5 | packaged by conda-forge | (default, Jul 31 2020, 02:25:36) 
[Clang 10.0.1 ] on darwin
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',20)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> ### Begin command Std_Workbench
>>> Gui.activateWorkbench("DraftWorkbench")
>>> ### End command Std_Workbench
>>> MRU="0"
>>> exec(open('/Applications/FreeCAD.app/Contents/Resources/share/Mod/Start/StartPage/LoadMRU.py').read())
>>> FreeCAD.openDocument('/Users/Mom and Dads house.FCStd')
>>> # App.setActiveDocument("Mom_and_Dads_house")
>>> # App.ActiveDocument=App.getDocument("Mom_and_Dads_house")
>>> # Gui.ActiveDocument=Gui.getDocument("Mom_and_Dads_house")
>>> Gui.runCommand('Std_OrthographicCamera',1)
>>> Gui.runCommand('Draft_Line',0)
>>> import Draft
>>> pl = FreeCAD.Placement()
>>> pl.Rotation.Q = (0.0, 0.0, 0.0, 1.0)
>>> pl.Base = FreeCAD.Vector(0.0, 2.0, 0.0)
>>> points = [FreeCAD.Vector(0.0, 2.0, 0.0), FreeCAD.Vector(10.0, 2.0, 0.0)]
>>> line = Draft.makeWire(points, placement=pl, closed=False, face=True, support=None)
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line')
>>> Draft.autogroup(line)
>>> FreeCAD.ActiveDocument.recompute()
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line','Vertex2',10,2,0)
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line','Edge1',5.67005,2,0)
>>> FreeCAD.getDocument('Mom_and_Dads_house').getObject('Line').Length = '25.4 mm'
>>> 
>>> FreeCAD.getDocument('Mom_and_Dads_house').getObject('Line').Length = '7.62 mm'
>>> 
>>> FreeCAD.getDocument('Mom_and_Dads_house').getObject('Line').Length = '8.89 mm'
>>> 
>>> Gui.runCommand('Std_WhatsThis',0)
>>> Gui.runCommand('Std_About',0)
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line','',1.91164,2,0)
>>> ### Begin command Std_Workbench
>>> Gui.activateWorkbench("ArchWorkbench")
>>> ### End command Std_Workbench
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 7.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 8.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 7.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 6.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 5.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 4.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 3.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 4.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 5.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 6.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 7.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 8.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 9.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 10.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 11.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 10.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 9.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 8.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 7.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 6.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 5.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 6.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 7.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 8.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 9.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 10.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 11.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 12.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 13.00
>>> 
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line')
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line')
>>> Gui.runCommand('Std_DlgCustomize',0)
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 7.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 6.00
>>> 
>>> FreeCADGui.getDocument('Mom_and_Dads_house').getObject('Line').LineWidth = 9.00
>>> 
>>> App.closeDocument("Mom_and_Dads_house")
>>> # App.setActiveDocument("")
>>> # App.ActiveDocument=None
>>> # Gui.ActiveDocument=None
>>> # Gui.Selection.clearSelection()
>>> MRU="0"
>>> exec(open('/Applications/FreeCAD.app/Contents/Resources/share/Mod/Start/StartPage/LoadMRU.py').read())
>>> FreeCAD.openDocument('/Users/Mom and Dads house.FCStd')
>>> # App.setActiveDocument("Mom_and_Dads_house")
>>> # App.ActiveDocument=App.getDocument("Mom_and_Dads_house")
>>> # Gui.ActiveDocument=Gui.getDocument("Mom_and_Dads_house")
>>> Gui.runCommand("Draft_ToggleGrid")
>>> Gui.runCommand('Draft_Line',0)
>>> pl = FreeCAD.Placement()
>>> pl.Rotation.Q = (0.0, 0.0, 0.0, 1.0)
>>> pl.Base = FreeCAD.Vector(0.0, 10.0, 0.0)
>>> points = [FreeCAD.Vector(0.0, 10.0, 0.0), FreeCAD.Vector(20.0, 10.0, 0.0)]
>>> line = Draft.makeWire(points, placement=pl, closed=False, face=True, support=None)
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line')
>>> Draft.autogroup(line)
>>> FreeCAD.ActiveDocument.recompute()
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Mom_and_Dads_house','Line','Vertex2',20,10,0)
>>> Gui.runCommand('Std_Copy',0)
>>> 

# Steps to reproduce

1. click the create a 2 lines tool. 
2. Draw a straight line
3. Make sure line is selected
4. In toolbar try and increment the width of the line past 7px or on view tab click line width and increment it. 
 

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22246 (Git)
Build type: Release
Branch: master
Hash: abd18ff88f7682e9b1160c36bb785b14a33473ef
Python version: 3.8.5
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Catalina 10.15.5**
* **Platform:** macOS
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2020-08-21 20:43
Confirmed on the forum thread

luzpaz, 2022-04-12:
Looks like @PLChris is having issues per the [forum thread](https://forum.freecadweb.org/viewtopic.php?p=573688#p573688). @ipatch + @chennes do you mind testing on macOS and confirming ?

ipatch, 2022-04-12:
using the [build](https://github.com/FreeCAD/homebrew-freecad/releases/tag/0.20.28575-git-master) of freecad i published a around a week ago. and a recent build i've been using,

```
[code]
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.20.28638 (Git)
Build type: Release
Branch: ipatch.builds.master.head.mac.app
Hash: dc3986ee1083e6ad0a8fcc11b00ebec6f8966638
Python 3.9.12, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * Behave-Dark-Colors 0.0.1
  * Plot
  * CfdOF
  * Defeaturing
  * CfdOF.bak 1.9.6
  * A2plus 0.4.56
  * Manipulator 1.4.3
  * fasteners 0.3.40
[/code]
```

it looks like the appearance of a line in the **Draft** workbench does not increase ie. there is no visual difference in the viewport with the line size is increased beyond **10**

[image]

luzpaz, 2022-04-12:
@donovaly should we fix before release v0.20 release?

donovaly, 2022-04-13:
> @donovaly should we fix before release v0.20 release?

No crash, no regression, therefore me will have to postpone this - except someone sends in a patch of course.

Inviting @marioalexis84 and @Roy-043 .

donovaly, 2022-04-13:
To me it seems it is a bug in Draft, not in Arch, right?

marioalexis84, 2022-04-13:
It's a bug in Coin. I can reproduce use only Coin in a very basic example with no relation to FreeCAD.
In the other hand,  seems to be platform dependent: After trying in several machines to draw a basic cube in FreeCAD and change the line width, only works fine in the older machine.
Attempting to downgrade the OpenGL implementation to a version earlier than the one used on the machine where the error did not occur was unsuccessful.

luzpaz, 2022-04-13:
@marioalexis84 thanks for the research. Do you mind opening an upstream ticket on https://github.com/coin3d/coin/issues ?

WandererFan, 2022-04-13:
Line width change not reflected in display is one of the symptoms in #6698.  Coin uses Quarter widget which is based on Qt Graphics Framework as is page display in TechDraw.  Any chance the issues are related?

marioalexis84, 2022-04-13:
I don't see #6698 in TechDraw with Qt5.15.2 nor Qt5.15.3, 
For completeness, just add that VTK also uses Qt+OpenGL and there is no issue with the line width in the scene.

luzpaz, 2022-04-29:
@marioalexis84 is there an upstream ticket yet?

marioalexis84, 2022-04-29:
Not yet.

marioalexis84, 2022-06-07:
> do you mind opening an upstream ticket?

I will try in the next days.

luzpaz, 2023-06-01:
Any progress ?

maxwxyz, 2024-08-12:
is this still relevant?

luzpaz, 2024-10-31:
Can you test on weekly (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) ?

goleon69, 2024-11-01:
Done and it's the same.
Not sure can help but on my Mac which is a little bit old (12.7), I cannot install the latest Blender due to changes in graphics libraries that works only with the newest macOS.

OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git)
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * IconThemes.backup1670063478.460454
  * ExtremeProDark.backup1670063426.751586
  * Glass.backup1670063844.392323
  * PieMenu.backup1670064528.67621 1.1.10

FEA-eng, 2025-09-14:
For me, the limit when it doesn't become thicker anymore is 10 instead of 7.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2026-09-20:
> For me, the limit when it doesn't become thicker anymore is 10 instead of 7.

@FEA-eng does this mean that it's not macOS exclusive ?

FEA-eng, 2026-09-20:
> [@FEA-eng](https://github.com/FEA-eng) does this mean that it's not macOS exclusive ?

On Windows, the limit seems to be higher, but there's still a value (10) after which the line is not getting any bigger. Perhaps it's intentional, though (to avoid absurdly thick lines).

flaviut, 2026-11-02:
Blocked on https://github.com/coin3d/coin/issues/839