FreeCAD/FreeCAD #6102: qwindowsvistastyle.dll does not style the line color

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4423

* **Reporter:** Roy_043
* **Date submitted:** 8/25/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** UI

# Original report text

The lines (frames) in task panels and dialog boxes are not styled by qwindowsvistastyle.dll but the background is. In recent Windows version the default line color is light gray. This color is almost indistinguishable from the light blue background color.

# Additional information

Related Forum topics:
https://forum.freecadweb.org/viewtopic.php?f=3&t=49191
https://forum.freecadweb.org/viewtopic.php?f=3&t=45022
https://forum.freecadweb.org/viewtopic.php?f=21&t=49174 (where the issue has been confirmed)

# Steps to reproduce

1. Start any command that shows a task panel or dialog box.
2. The frame lines in the task panel/dialog box are almost invisible.
3. Exit FreeCAD.
4. Rename \bin\styles\qwindowsvistastyle.dll
5. Restart FreeCAD.
6. Start any command that shows a task panel or dialog box.
7. The frame lines in the task panel/dialog box are visible now.
8. But the this workaround has a negative side-effect: Selected option, f.e. in the Draft Snap toolbar are harder to distinguish.

# FreeCAD Info

```
OS: Windows 8.1 (6.3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22262 (Git)
Build type: Release
Branch: master
Hash: f56825c79f11b75648a7a0dcc8a9f52bb4574e61
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Dutch/Netherlands (nl_NL)

```

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **OS: Windows **
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Roy_043 2020-08-25 08:47
Oops: the summary should be: qwindowsvistastyle.dll does NOT style the line color

---

### Comment by Roy_043 2020-08-25 09:11
Oops 2: the DLL that should be renamed in step 4 is \bin\styles\qwindowsvistastyle.dll

---

### Comment by Kunda1 2020-09-03 21:52
revised summary

luzpaz, 2022-03-13:
CC @Roy-043 (OP)

luzpaz, 2022-04-01:
@Roy-043 is this still valid?

Roy-043, 2022-04-02:
Yes, the problem still exists.

```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.20.28193 (Git)
Build type: Release
Python 3.8.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
```

luzpaz, 2022-04-15:
@Roy-043 can you provide a succinct screenshot of the issue to add to this ticket? I think that would help get this ticket addressed

Roy-043, 2022-04-16:
[image]

On the left a task panel when `qwindowsvistastyle.dll` is available. On the right the same task panel after `qwindowsvistastyle.dll` has been renamed.

luzpaz, 2022-04-16:
@donovaly do we want to try to get this in before v0.20 ?

donovaly, 2022-04-18:
Does the bug only occur when qwindowsvistastyle.dll is missing? I checked and this file is part of our Windows releases.

donovaly, 2022-04-19:
> You have misunderstood. In qwindowsvistastyle.dll the frames are not styled properly.

 qwindowsvistastyle.dll is part of Qt. The image at the left in your post: https://github.com/FreeCAD/FreeCAD/issues/6102#issuecomment-1100674245
looks right, the one at the right uses Win XP style.

If you like to have the QGroupBox frame in another color, this can be changed. This is just how we define it in our default style sheet. Take for example the Style "light-orange" in the FreeCAD settings and you get a dark-gray frame.

Therefore I propose you make a PR for FC's default *.css file to use another color. This can go  in for FC 0.20. If you like, open a new issue for this.

 Since the Qt DLL is not to blame - > closing this issue to keep the overview

Roy-043, 2022-04-19:
I suppose you mean *.qss files? Can you tell me where the default style sheet is located? I can only find the custom style sheets (...\data\Gui\Stylesheets\). Thanks.

donovaly, 2022-04-19:
> I suppose you mean *.qss files?

Yes.

> Can you tell me where the default style sheet is located?

I am in vacation and thus away from my development PCs. If you ask in the developer's forum, someone can tell you where this is. Since our task panels use a non-standard color design, I am sure we just need to change a color setting somewhere in our code.

luzpaz, 2022-04-19:
lol @donovaly make sure you vacation some (so you don't burn out).

donovaly, 2022-07-29:
Reopening in request here: https://forum.freecadweb.org/viewtopic.php?p=612793#p612793

Roy-043, 2022-07-29:
Apparently the pale blue task panel background color is not styled by the mentioned DLL but by the FreeCAD code. Choosing a different color, that does not hide the frames, would solve the issue then.

donovaly, 2022-07-31:
> Apparently the pale blue task panel background color is not styled by the mentioned DLL but by the FreeCAD code. Choosing a different color, that does not hide the frames, would solve the issue then.

So the frame itself is styled by Windows?
If so, then we need a discussion in the forum. I expect many users to keep the current color scheme as default.
If not, do you know where in the code the frame is colored?

Roy-043, 2022-07-31:
The light blue background color is `#d6dff7`.
The light gray frame color is `#dddddd`.

Looking for `#d6dff7` in the source code gives me two files:

https://github.com/FreeCAD/FreeCAD/blob/b40de7a509b83627dcc9ead2c1ea3a91343938ae/src/Gui/QSint/actionpanel/freecadscheme.cpp#L63-L66

https://github.com/FreeCAD/FreeCAD/blob/b40de7a509b83627dcc9ead2c1ea3a91343938ae/src/Gui/QSint/actionpanel/winxppanelscheme.cpp#L99-L102

The specified border color (`#ffffff` = white) is somehow overwritten. I think `qwindowsvistastyle.dll` is responsible for that. See the image below (already posted before). In the screenshot on the right, where `qwindowsvistastyle.dll` was renamed, the frames are white.

[image]

donovaly, 2022-07-31:
> Looking for `#d6dff7` in the source code gives me two files:
> 
> https://github.com/FreeCAD/FreeCAD/blob/b40de7a509b83627dcc9ead2c1ea3a91343938ae/src/Gui/QSint/actionpanel/freecadscheme.cpp#L63-L66

Thanks. However, the border there is not the groupbox frame. When I set the width of both frames defined in the file to 10 px I get this:
[image]

Searching for #dddddd brings me no results. So I have no idea where the groupbox color is set.

donovaly, 2022-07-31:
OK, I found the reason: we don't set the groupbox at all, therefore the Windows style is applied.

The solution is to set it to a color that suits us. I have done this: #7292

PR #7292 (2022-08-01): [Gui] QSint: fix QGroupBox visibility
- fixes #6102
- the point is that we don't set any style for the group boxes, therefore the windows style steps as nothing was set yet. Its default style is the color #dddddd which is hardly visible on our blue default style.
as solution set an explicit color and use the grooved style.

This is the result:
[image]
commit 360d1336 (2022-08-01): - fixes #6102
- the point is that we don't set any style for the group boxes, therefore the Windows style is applied as nothing was set yet. The default Windows style is the color #dddddd which is hardly visible on our blue default style.
as solution set an explicit color.
commit 6146311a (2022-08-02): - fixes #6102
- the point is that we don't set any style for the group boxes, therefore the Windows style is applied as nothing was set yet. The default Windows style is the color #dddddd which is hardly visible on our blue default style.
as solution set an explicit color.
closed 2022-08-01 by donovaly