FreeCAD/FreeCAD #6103: Crash after editing Part Design feature links with DAGView active

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4431

* **Reporter:** Gregory son of Carl
* **Date submitted:** 9/9/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #pending-forum, crash, DAG, Part Design

# Original report text

[Link to forum thread](https://forum.freecadweb.org/viewtopic.php?f=20&t=49982)

When the DAGView panel is active, FreeCAD will crash after editing App:Link objects that reference Part Design features such as Pads, Revolves, Sweeps, and Sketches.

# Additional information

```FreeCAD 0.19, Libs: 0.19R22284 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Loading Assembly4 WorkBench
Sheet Metal workbench loaded
During initialization the error "'backend.qt4 is not a valid rc parameter (see rcParams.keys() for a list of valid parameters)'" occurred in /home/gregory/.FreeCAD/Mod/animation/InitGui.py
Please look into the log file for further information
During initialization the error "inconsistent use of tabs and spaces in indentation (<string>, line 48)" occurred in /home/gregory/.FreeCAD/Mod/GDT/InitGui.py
Please look into the log file for further information
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7fd13613f840]
#1  0x7fd138ba2f54 in Gui::DAG::Model::slotResetEdit(Gui::ViewProviderDocumentObject const&) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x44
#2  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so(+0x382d1c) [0x7fd13892ad1c]
#3  0x7fd138981c35 in Gui::Document::_resetEdit() from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x1f5
#4  0x7fd13891022a in Gui::Application::setEditDocument(Gui::Document*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x3a
#5  0x7fd13899177c in Gui::DocumentPy::resetEdit(_object*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x2c
#6  0x7fd1389917c4 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x24
#7  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(PyCFunction_Call+0xf7) [0x7fd138393d87]
#8  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x250) [0x7fd138348850]
#9  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4e36) [0x7fd1383d77d6]
#10  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x2e2) [0x7fd138344f92]
#11  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x44) [0x7fd138345d54]
#12  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCode+0x1c) [0x7fd1383fb6ac]
#13  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(+0x219ecf) [0x7fd138419ecf]
#14  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(+0x252b84) [0x7fd138452b84]
#15  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libpython3.8.so.1.0(PyRun_StringFlags+0x7d) [0x7fd138452c1d]
#16  0x7fd137c4d2ba in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADBase.so+0x6a
#17  0x7fd1389ec3b8 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x98
#18  0x7fd1389ec5a2 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x102
#19  0x7fd04a7a9e35 in PartDesignGui::TaskDlgFeatureParameters::reject() from /tmp/.mount_FreeCAbs6iBc/usr/lib/PartDesignGui.so+0x105
#20  0x7fd04a7aeaa1 in PartDesignGui::TaskDlgSketchBasedParameters::reject() from /tmp/.mount_FreeCAbs6iBc/usr/lib/PartDesignGui.so+0x31
#21  0x7fd138c85819 in Gui::TaskView::TaskView::reject() from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x59
#22  0x7fd1366867b8 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Core.so.5+0x780
#23  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5(+0x274bbb) [0x7fd136f9cbbb]
#24  0x7fd1366867b8 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Core.so.5+0x780
#25  0x7fd136f351fe in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5+0x32
#26  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5(+0x20d8b7) [0x7fd136f358b7]
#27  /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5(+0x20e14c) [0x7fd136f3614c]
#28  0x7fd136f361e8 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5+0x88
#29  0x7fd136ec7c3e in QWidget::event(QEvent*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5+0xce
#30  0x7fd136e9c82a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5+0xde
#31  0x7fd136ea1095 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Widgets.so.5+0x79d
#32  0x7fd1389baab9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libFreeCADGui.so+0x89
#33  0x7fd136670186 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAbs6iBc/usr/bin/../lib/libQt5Core.so.5+0x88
[... 47 lines cut ...]
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 2/4/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2020-09-10 14:49
Confirmed by chrisb on the thread

---

### Comment by Kunda1 2020-12-04 02:51
chennes responded in https://forum.freecadweb.org/viewtopic.php?f=20&t=49982#p454170

---

### Comment by Kunda1 2022-02-04 19:00
bumped forum thread

luzpaz, 2022-04-15:
@donovaly this is a crash bug but haven't been able to reproduce. But not sure I'm doing it right

luzpaz, 2022-04-15:
@realthunder can you reproduce?

donovaly, 2022-04-16:
As long as nobody can reproduce this with current FreeCAD master, I see no need for action.

I cannot reproduce, in the forum I see @kisolre reports the same.

In case it is reproducible, please reopen.

closed 2022-04-16 by donovaly