FreeCAD/FreeCAD #6105: Flippin' direction of line / curve

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4438

* **Reporter:** Pauvres_honteux
* **Date submitted:** 9/18/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

While discussing https://forum.freecadweb.org/viewtopic.php?f=34&t=44278&start=10&sid=6e87795aa14b31f3b0ce47060b5f0f7e the denomination of the command "Reverting shape" it came to my attention that flippin' the direction of a line or curve was missing.
Therefore I hereby request it to be added to "Reverting shape".

I also request a name change of "Reverting shape" to "flip normal/direction".

And finally I request a new icon in the tree somehow involving an arrow of some sort.
And a clickable arrow in the 3D-viewport attached to the marked line/curve/plane/surface.

# Steps to reproduce

Doesn't exist.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Pauvres_honteux 2020-09-18 11:19
... this homepage is not mobile friendly at all...  =O(

maxwxyz, 2024-09-01:
@Roy-043 do you know if the flipping is meant for draft or sketcher lines or just a mirror object?

Roy-043, 2024-09-02:
I am not sure where 'mirror' comes from, Part_ReverseShape flips the face normals of an object. All normals are flipped, which may not always be what is required. For sketches you would most likely want a solution that can reverse selected edges (#11957). And probably not need/want a parametric solution.

I personally don't see much in extending the Part_ReverseShape tool as requested.

maxwxyz, 2024-09-04:
I'll close this then.

closed 2024-09-04 by maxwxyz