FreeCAD/FreeCAD #6106: Safe Mode (ala LibreOffice) that offers comprehensive options to deal with user config files

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4443

* **Reporter:** Kunda1
* **Date submitted:** 9/25/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #ready-for-migration, UI

# Original report text

Apparently LO has a cool feature that helps users deal with corrupt profile/config files. It's called Safe Mode. It backs up the current config files and uses a vanilla copy. This would be mean: No more asking users to back up their ~/.FreeCAD/ and ~/.config/FreeCAD directories (well at least the majority of usecases).

[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=8&t=50342)

# Additional information

!(https://forum.freecadweb.org/download/file.php?id=127630)

# Other bug information

* **Priority:** none
* **Severity:** feature
* **Category:** Feature
* **Updated:** 1/25/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2021-02-13 15:05
Helpful findConfigFiles macro by TheMarkster that may be useful when creating this feature
https://forum.freecadweb.org/viewtopic.php?f=22&t=29888

---

### Comment by Kunda1 2021-03-09 00:16
https://github.com/furti/FreeCAD-PreferencesPresets

adrianinsaval, 2023-01-18:
This could also use software OpenGL so that this can be easily tested

luzpaz, 2023-09-24:
This feature code could also be utilized in #5772

luzpaz, 2023-10-09:
This special mode can also be a way to start FreeCAD without 3rd party addons, so users and testers can help troubleshoot bugs whose usecase is related to an installed external addon.

GAZ082, 2024-09-12:
This would be a nice feature.

JohnOCFII, 2024-09-12:
Great idea.  OctoPrint, the 3D printing software tool also has a similar safe-mode feature.

luzpaz, 2024-09-13:
Another usecase: 
* https://github.com/FreeCAD/FreeCAD/issues/16478#issuecomment-2346562289

GAZ082, 2024-09-13:
> Another usecase:
> 
> * [Windows: Crash 'Access Violation' (1.0 rc1) #16478 (comment)](https://github.com/FreeCAD/FreeCAD/issues/16478#issuecomment-2346562289)

Thank you @luzpaz for making a case for this feature. We have a sizeable amount of issues triggered by "something" in the configs, which is really annoying.

luzpaz, 2024-09-14:
> 20/20 hindsight I should have have somehow saved my broken config before resetting... 😕

Source: https://github.com/FreeCAD/FreeCAD/issues/16525#issuecomment-2351110671

hyarion, 2024-09-17:
How about this:

* if freecad fails to start, it writes a special `FREECAD_SAFE_MODE` file to /tmp (or similar dir).
* on a successful load, it removes the special safe-mode file.
* if there's a safe-mode file available or if freecad is started with a new flag `--safe-mode`, then freecad loads into safe-mode.

Safe-mode could be the following:
* disable addons installed by user (skip the mod location that addon manager installs files into)
* avoid loading _and_ saving config files

Basically try to load freecad in a factory default state.

Would this be enough? Or would you like to change anything for a first iteration?

---

In a later iteration on this, we could extended it to load a special dialog to let the user preform different actions, similar to the one from the screenshot in the ticket.

GAZ082, 2024-09-17:
@hyarion 
Hi Ben. When you refer to avoid loading a config file, you basically mean skipping all user modifications. I'm not very well versed on how configs work in FC, sorry if this seems to be a stupid question.

luzpaz, 2024-09-17:
Hi @wwmayer, could you weigh in ? :pray: (https://github.com/FreeCAD/FreeCAD/issues/6106#issuecomment-2356931221)

hyarion, 2024-09-18:
> When you refer to avoid loading a config file, you basically mean skipping all user modifications. 

Config file contains all settings, addons means plugins.
But the end result would be that we skip all user modifications so we get a clean state.

luzpaz, 2024-09-20:
@Chennes do you have a moment to weigh-in on the proposal https://github.com/FreeCAD/FreeCAD/issues/6106#issuecomment-2356931221 ?

chennes, 2024-09-20:
This seems fine to me. Can we also disable Qt's Settings store?

luzpaz, 2024-09-20:
> Can we also disable Qt's Settings store?  

@hyarion ?

hyarion, 2024-09-20:
Sure!

luzpaz, 2024-09-20:
@hyarion can I assign you to this ticket ?

hyarion, 2024-09-20:
@luzpaz feel free to assign it to me :)

hyarion, 2024-09-20:
@chennes: `QSettings` seems a bit difficult to disable as it is used for a lot of different things.
For example, we still want it to be able to load ini files for us and get windows version from the windows registry.

We could let a new `FCSettings` subclass `QSettings` and let that handle if it should pass on the calls to super or just keeping the settings in memory during the session.
I do feel that this might be more complicated than what is worth.

Do you (or anyone else?) know why some code stores settings in `QSettings` while other code stores similar data as parameters in `user.cfg` using `App::GetApplication().GetParameterGroupByPath()` and similar?

Would it be an option to refactor the settings stored in QSettings to store the data in `user.cfg` instead?

chennes, 2024-09-21:
`QSettings` is mostly used to store the stuff that Qt wants to store there (toolbar positions, window positions, etc.) while it would be delightful to rip it out entirely, that is a project in its own right. I'm happy with the new Safe Mode not touching it for the time being, it was a suggestions of a "nice to have" feature, not an essential component.

hyarion, 2024-09-22:
I have a first implementation of the features discussed above

The implementation can be found in this branch: https://github.com/hyarion/FreeCAD/tree/safe-mode

Here is what it does:
* Creates a new temporary directory which is removed when the user exit freecad
* The following directories are mapped to new directories in the newly created temporary directory: `UserAppData`, `UserConfigPath`, `UserCachePath`, `AppTempPath`, `UserMacroPath`, `UserHomePath`

Safe mode can be entered in three ways:
* When `--safe-mode` is added as an argument when starting the program
* From menu -> help -> restart in safe mode
* If freecad crashed the last time (in the last x hours)

When start mode is enabled, it will appends ` (Safe Mode)` in the main window title and show a pop up after start informing the user that safe mode is entered.

What's missing is:
* QSettings (nice to have, can be added later)
* UI for configuration like in libreoffice, though I think we need to refine what we actually want before that can be implemented.
* Icon for command in help menu, reuses the `view-refresh` icon for now.

Caveats:
* The command in help is still enabled in safe mode, making it possible to restart into safe mode, from safe mode. (I haven't decided if this is a feature or bug)
* If the user preforms a change in safe mode that requires freecad to restart, it will restart in safe mode again, with the changes wiped (I'm not sure if this is a feature or bug either)

Screenshots:

Help menu:
[image]

Popup after load ([qt on mac ignores the window title on dialogs, as required by apples guide lines](https://doc.qt.io/qt-6/qmessagebox.html#setWindowTitle)):
[image]

Window title:
[image]

What do you think, is this in line of what has been requested? Is there anything you want to be changed or added to the first iteration?
I will wait for feedback here before creating a merge request.

luzpaz, 2024-09-22:
This is a huge first step! 
:wave: @pgilfernandez, do you have time to make a Safe Mode icon ? 
[image]

GAZ082, 2024-09-22:
@luzpaz i could just change the color of the current icon to something like green real quick

GAZ082, 2024-09-22:
[image]

pgilfernandez, 2024-09-23:
Hi!,
Is enough with that color change or do you prefer to design a whole new one? I would do it if you need it

hyarion, 2024-09-23:
> Is enough with that color change or do you prefer to design a whole new one? I would do it if you need it

I think the current one is good enough for the current version. But if you want to design one then I'll happily include it 😃

luzpaz, 2024-09-23:
@GAZ082 IMHO, I think the choice of color, green, doesn't indicate to me of a 'safe mode' state.  Yellow or red seem more appropriate. 

@pgilfernandez maybe a new one ? :stuck_out_tongue_winking_eye:

GAZ082, 2024-09-23:
@luzpaz i made it green because you know, safe? Red is bad! 😁

JohnOCFII, 2024-09-23:
> @luzpaz i made it green because you know, safe? Red is bad! 😁
[image]

maybe add a small green cross like the pharmacies in Europe

GAZ082, 2024-09-23:
@JohnOCFII I thought about that until you realize that the cross is mostly a western thing in the Middle East you have the crescent instead.

luzpaz, 2024-09-23:
[image]

hyarion, 2024-09-23:
[image]

adrianinsaval, 2024-09-23:
for icon choices please ask @FreeCAD/design-working-group

hyarion, 2024-09-23:
@adrianinsaval how and where do I ask them?

adrianinsaval, 2024-09-24:
by pinging the group as I did or on the discussion in the discord server https://discord.gg/w2cTKGzccC

GAZ082, 2024-09-24:
asking as we speak

obelisk79, 2024-09-24:
[image]

GAZ082, 2024-09-24:
@obelisk79 not a fan of these as a cross may be widely recognized but it's not used in the middle east.

MisterMakerNL, 2024-09-24:
After a quick search I find this icon a lot:
[image]

pgilfernandez, 2024-09-24:
> After a quick search I find this icon a lot:
> 
> [image]
> 
> 

Yes, this is the idea I was going to design, a green shield...

hyarion, 2024-09-24:
> ![rescue2](https://private-user-images.githubusercontent.com/17993710/370108400-f3d6176f-b137-4e26-8ee2-e1841ef5e8e1.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNjgzMzQsIm5iZiI6MTcyNzE2ODAzNCwicGF0aCI6Ii8xNzk5MzcxMC8zNzAxMDg0MDAtZjNkNjE3NmYtYjEzNy00ZTI2LThlZTItZTE4NDFlZjVlOGUxLnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDA4NTM1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQwZDZlYTFmYzhkOWJkMGFjOTEyODZhYWVkMDk5NjZiNGU0MDZhYTE5OWNkOTFjYjRiNzJjZWFjMDc5MDQ3ZTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7Y_VUjLyxTN5JtVZO9ox6gIpcazg5jdjJgKCrZbR_Ow)

The red cross on a white background, ie. the reverse of the swiss flag, is protected by the geneva convention. Let's not violate it :)

hyarion, 2024-09-24:
Thanks for all the input on icons, I really appreciate it!

I think many of them represent safe mode quite well, but in this case we primarily need an icon for the command/action to "restart in safe mode". Not for "Safe Mode" as a concept (though that could be added to the dialog after the restart).

Before deciding on an icon, please check the [comment I made with the screenshots](https://github.com/FreeCAD/FreeCAD/issues/6106#issuecomment-2366744455) that shows the context of where it will be used.

GAZ082, 2024-09-24:
Easy peasy, lets do the green shield but replace the tick with the circled arrows of the restart icon.

GAZ082, 2024-09-24:
Something like this:

[image]

tobiasfalk, 2024-09-24:
> Something like this:
> 
> [image]
> 

Could you make the arrows blue, like the original restart icon

GAZ082, 2024-09-24:
That would be kinda ugly, but I can try. If you guys want I can do something tonight after kids go to bed and if people likes it, make a PR.

GAZ082, 2024-09-24:
[image]

Like?

hyarion, 2024-09-24:
Scaling is handled automatically as freecad uses svg icons.

I'm not at designer so I might be the wrong person to give feedback here, but I'll try anyway ^^

I think the icon looks quite nice, but I think it doesn't quite follow [the artwork guidelines](https://wiki.freecad.org/Artwork_Guidelines), does it?

* The colors doesn't match with those in the green palette.
[image]
* There's also a highlight missing right inside of the outline.
* I think a white arrows should outline should be the brightest white. Then the fill should probably be a gradient with the gray you to the outline color, similar to how you have the fill for the shield.

GAZ082, 2024-09-24:
Followed the guidelines. Had to use the Tango pallete as a base because the FC one is not available (do not have access to the GDrive file).

I'll see if can do more changes today!

hyarion, 2024-09-26:
@luzpaz https://github.com/FreeCAD/FreeCAD/issues/16834 another usecase

luzpaz, 2024-09-26:
@hyarion maybe we should just merge to main as is and continue updating the functionality/esthetics as we go ?

GAZ082, 2024-09-26:
I got access to the FC pallete, will have a look at it today and do a review. Then see if can add a PR to your brach @hyarion , what you say?

Just to get something started, then i can refine the icon (or someone else).

hyarion, 2024-09-26:
Sure go for it!
If we don't manage to get the icon ready in time (not sure how icons are reviewed), we can always add the icon in the next iteration of the tool :)

GAZ082, 2024-09-26:
@obelisk79 is there any formal icon review process?

GAZ082, 2024-09-26:
@hyarion check this out 
[image]

hyarion, 2024-09-26:
Thanks for taking the time, I like the icon!

For the record, @GAZ082 popped in the discord channel so if anyone wants to help, feel free to pop by the design-working-group channel :)

luzpaz, 2024-09-26:
> If we remove that feature, the only ways that safe-mode can get activated is manually through the help menu and by adding `--safe-mode` on the command line - and that should be safe to merge.

Yes, this is a smart compromise, imho

GAZ082, 2024-09-26:
The guys at the discord toned down the gradient:

[image]

hyarion, 2024-09-26:
Cool, great work!

luzpaz, 2024-09-26:
Team work makes the dream work :fist_right: :fist_left:

[14 more replies]