FreeCAD/FreeCAD #6108: Fullscreen renders nothing

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4446

* **Reporter:** qwerty_f
* **Date submitted:** 9/27/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** #pending-forum, GUI, macOS

# Original report text

Since I randomly hit the touchbar turning on full screen FreeCAD switches to full screen mode. Noting except for the sides-cube and the axis icon is rendered. Nothing else. Closing the screen returns to an empty window. Only escape is to close FreeCAD and start over.

# Steps to reproduce

(Intentionally) Hit the hot key for full screen (on my machines it's F11).

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22474 (Git)
Build type: Release
Branch: master
Hash: a44f8ffd427fa9b23b1f00dbf62d66cd152cd774
Python version: 3.8.5
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Catalina 10.15.6 (19G2021**
* **Platform:** Mac
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2020-10-09 21:24
qwerty_f : please address your issue according [our reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by opening a new topic on [our forum](https://forum.freecadweb.org/index.php). Thx
Consider searching for similar report in our forum, IIRC there already is a thread about this.

---

### Comment by qwerty_f 2020-10-09 21:39
openBrain will do so

---

### Comment by Kunda1 2020-10-10 12:03
vejmarie can you confirm?

---

### Comment by Kunda1 2020-10-15 14:11
Discussion thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=51187

---

### Comment by qwerty_f 2020-10-15 15:33
Uhm, I already entered into a similar thread about full screen to see what the outcome would be. Nothing so far, so another one seems superfluous. Sorry for the confusion.

---

### Comment by Kunda1 2020-10-15 15:50
qwerty_f this is bugtracker that tracks the progress of bugs/features. If you don't post the forum thread associated with the ticket as openBrain pointed out in ~0014809 then it's only superfluous to you and not the rest of us reading this ticket.

---

### Comment by Kunda1 2020-10-15 15:50
chrisb can you confirm issue on macOS?

luzpaz, 2022-04-01:
@PLChris @chennes can you reproduce on macOS?

JohnOCFII, 2022-04-03:
> 

I'm not them, but I don't see this issue with any of: the last weekly: 27936; nor iPatch's 28574, nor my self-compiled (with Homebrew) 28578.

[code]
OS: macOS 12.3
Word size of FreeCAD: 64-bit
Version: 0.20.28578 (Git)
Build type: Release
Branch: master
Hash: 4cffe64b623054207050a047bbb1926a83ceb747
Python 3.9.12, Qt 5.15.3, Coin 4.0.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]

luzpaz, 2022-04-03:
Thanks! Closing. If this issue persists, please don't hesitate to re-open.

closed 2022-04-03 by luzpaz