FreeCAD/FreeCAD #6109: 0.19 Tech Draw does not correctly open files from previous versions.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4448

* **Reporter:** CrashedAgain
* **Date submitted:** 10/3/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** resolved
* **Tags:** 0.18, 0.19, TechDraw

# Original report text

0.19 Tech Draw does not correctly interpret the rotation angle of any rotated views created by previous versions. For example, if the page contains a view that has been rotated -90 degrees, 0.19 displays the view as being rotated 0 degrees. This causes any dimensions in the view to be incorrect (usually to "0.00") because the anchor points are shifted. 

# Additional information

The only additional information I have is that a file originally rotated -90 degrees becomes correct when the rotation angle is set to 0; and a file originally rotated 30 degrees becomes correct when the rotation angle is reset to 60. Maybe 0.18 is measuring the angles from horizontal while 0.19 is measuring from vertical???? 

# Steps to reproduce

Step 1: Create a object in an earlier version (eg 0.18) and insert views of it into a Tech Draw page.
Step 2: Either rotate one (or more) of the views or create a section view and rotate it. Angle does not matter but it should be obvious. Add some x and y dimensions to the rotated view. 
Step 3:Save and close.
Step 4: Open the file in 0.19. You will find that (even though the angle in the data box is unchanged), the rotated view is displayed at a different angle and consequently the dimensions have changed. 
Obviously, this corrupted file cannot be used, therefore I have marked this bug as severe. The file can be repaired by editing the rotation data in the properties box until the orientation is correct.
Sample file attached...

# Other bug information

* **Priority:** high
* **Severity:** major
* **Category:** Bug
* **Platform:** All
* **Updated:** 2/2/2022

# Discussion from Mantis ticket

### Comment by wandererfan 2022-02-02 20:20
V018 had issues with direction of rotation and section views.  Note that the -90 rotation in the sample file is counter clockwise in v018, but clockwise (as it should be) in v019. 

Also note the inconsistency between section direction (looking down) and section normal ((0,0,-1) which indicates looking up). 

It is unfortunate, but I don't think we can duplicate errors in order to maintain backwards compatibility.

luzpaz, 2022-02-22:
@WandererFan nothing we can do for this? Not even a small migration script that could fix the CW-CCW issues?

WandererFan, 2022-02-23:
In the sample file, changing the Section's Rotation to 0 and recomputing restores the dimensions.   I guess that could be a script.

luzpaz, 2022-02-23:
@WandererFan and idea: start collecting bugs like this one and creating a migration script to make older TD files accessible?

WandererFan, 2022-02-24:
> @WandererFan <https://github.com/WandererFan> an idea: start collecting

Are there other bugs like this?

luzpaz, 2022-02-24:
I don't see them on the tracker. Perhaps I should open a ticket on the TD subdforum to invite users to contribute other bugs they've encountered?

WandererFan, 2022-02-24:
Users don't seem shy about reporting bugs, but go ahead. :)

WandererFan, 2022-02-26:
@luzpaz 
I have a macro that sorts out the Rotation changes from v0.18 to v0.19, but I don't know where to put it.  Is there a standard location for migration tools?
WF

luzpaz, 2022-02-26:
No idea. @yorikvanhavre any idea?

WandererFan, 2022-03-07:
Migration macro posted to FreeCAD/FreeCAD-macros/TechDraw/TechDrawMigrateFromv018.FCMacro.

closed 2022-03-17 by WandererFan