FreeCAD/FreeCAD #6110: When running FC 0.19 on Windows 7, it returns errors in the Report window

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4450

* **Reporter:** mozarelo
* **Date submitted:** 10/3/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** feedback
* **Tags:** #pending-forum

# Original report text

Each time it starts, it outputs:

[color=blue]23:00:50  OpenGL version is: 4.5 (4.5.0 NVIDIA 347.52)
23:00:50  Run Gui init script
23:00:51  Init: Running FreeCADGuiInit.py start script...
23:00:51  Init:   Searching modules...
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\AddonManager... done
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Arch... done
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Complete... done
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Draft... done
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Drawing... done
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Fem... done
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Idf(InitGui.py not found)... ignore
23:00:51  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Image... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Import... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Inspection... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Material... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Measure(InitGui.py not found)... ignore
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Mesh... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\MeshPart... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\OpenSCAD... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Part... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\PartDesign... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Path... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Points... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Raytracing... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\ReverseEngineering... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Robot... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Show(InitGui.py not found)... ignore
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Sketcher... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Spreadsheet... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Start... done
23:00:52  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Surface... done
23:00:53  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\TechDraw... done
23:00:53  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Test... done
23:00:53  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Tux... done
23:00:53  Init:      Initializing C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\Mod\Web... done
23:00:53  Module: Part
23:00:53  Loading Part module... done
23:00:53  Init:      Initializing C:\Users\Ultranemo\AppData\Roaming\FreeCAD\Mod\A2plus... failed
23:00:53  ----------------------------------------------------------------------------------------------------
23:00:53  Traceback (most recent call last):
  File "<string>", line 127, in InitApplications
  File "<string>", line 233, in <module>
  File "<string>", line 47, in __init__
  File "C:\Users\Ultranemo\AppData\Roaming\FreeCAD\Mod\A2plus\a2plib.py", line 35, in <module>
    import numpy
  File "C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\numpy\__init__.py", line 140, in <module>
    from . import core
  File "C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\numpy\core\__init__.py", line 101, in <module>
    from . import _internal
  File "C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\numpy\core\_internal.py", line 14, in <module>
    import ctypes
  File "C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\bin\lib\ctypes\__init__.py", line 468, in <module>
    GetLastError = windll.kernel32.GetLastError
  File "C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\bin\lib\ctypes\__init__.py", line 443, in __getattr__
    dll = self._dlltype(name)
  File "C:\FreeCAD_0.19.22522-Win-Conda_vc14.x-x86_64\bin\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
[... 192 lines cut ...]
---

### Comment by Kunda1 2020-10-15 14:13
Discussion thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=51188

---

### Comment by Kunda1 2020-10-15 16:01
mozarelo please follow this advice: https://forum.freecadweb.org/viewtopic.php?f=8&t=51188#p439657

---

### Comment by Kunda1 2020-10-15 18:43
mozarelo please make sure all your addons are up to date and try again. This includes A2+

---

### Comment by Kunda1 2021-03-28 16:24
mozarelo ping

---

### Comment by Kunda1 2021-03-28 16:27
Looks like these issues occur for mozarelo on win-conda and not on Win-LP
https://forum.freecadweb.org/viewtopic.php?f=8&t=51188#p439902

luzpaz, 2022-04-01:
bumped [forum thread](https://forum.freecadweb.org/viewtopic.php?p=585314#p585314)

adrianinsaval, 2022-04-02:
This looks like it's a python 3.8 incompatibility issue either with an older version of win7 (python 3.8 required that a specific windows update be applied to work) or with the processor they are using.

luzpaz, 2022-04-02:
GeneFC hasn't encountered the issue or incompatibility and thinks it may be a locale issue (russian). Anyway, closing ticket unless there are any changes (feel free to re-open)

closed 2022-04-02 by luzpaz