FreeCAD/FreeCAD #6114: annoying : confirmation popup window hidden behind main-window

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4459

* **Reporter:** anka
* **Date submitted:** 10/15/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** confirmed
* **Tags:** #pending

# Original report text

trying to add tool-controllers if more than one path-job exists, pops up a confirmation window for selecting the path-job.  
this selection ***window is not modal***, but hidden behind main window.

(seems to happen only in mac version)

see also : https://forum.freecadweb.org/viewtopic.php?f=3&t=51135

# Steps to reproduce

Gene gave a brief explanation that resumes best:

'I click the tool icon, my tool library pops up, I select a tool and create a controller. ***I am then asked which job gets the tool***.  
***The popup of last step is hidden behind main-window !!!***

There'e is an alternative (described by Chris): 
Select the pathjob **, edit it (double click), select 'Tools', then 'add',  tool library pops up, select a tool and create a controller
Doing it this way, I would find more logical if selected controller is created in the pathjob ** the procedure started with. For now, freecad also asks as final step to select a job.

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22670 (Git)
Build type: Release
Branch: master
Hash: 12155f4aab09047c5697db0c1b3cf93b02edda03
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: macosx  10.16.7  **
* **Platform:** mac
* **Updated:** 2/6/2021

sliptonic, 2022-03-07:
https://github.com/FreeCAD/FreeCAD/pull/6530

luzpaz, 2022-03-11:
Per merged #6530 the commit 182cabcfbbbe should fix this ticket. Closing.

closed 2022-03-11 by luzpaz