FreeCAD/FreeCAD #6117: FreeCAD Path needs a new Postprocessor architecture.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4468

* **Reporter:** sliptonic
* **Date submitted:** 10/18/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

The current post-processor architecture requires individual .py files for each post.  This is flexible in allowing users to copy an existing post and modify it but results in lots of duplicated code and redundant work.
Changes and improvements made in one post don't become part of the other posts so bugs and incompatibilities persist.

# Additional information

A better architecture would allow code reuse while preserving the users ability to customize the post to their needs.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

LarryWoestman, 2022-08-21:
I am currently working on refactoring the postprocessor code and the postprocessors.

I think it makes sense to assign this issue to me.

luzpaz, 2022-08-22:
@sliptonic :point_up:

luzpaz, 2024-05-08:
@sliptonic JFYI, this ticket is still open yet is labeled 'Done' in your CAM stuff list.

sliptonic, 2024-05-08:
Yeah, I left it open because the refactoring only touched the first part of post-processing.  The next step will be to start refactoring the individual posts to re-use code and to be easier to customize.

LarryWoestman, 2025-03-11:
Progress is being made on refactoring the individual posts, with more work to come.