FreeCAD/FreeCAD #6118: consistant unit usage

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4469

* **Reporter:** jaxxor
* **Date submitted:** 10/19/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** units

# Original report text

Hey all
Ugly american here to complain about the lack of support for the stupid foot/inch system.

In the image tool set, the option to scale the image only comes up in mm.
I am running the Appimage version of 0.19

anyway,
Thanks for any support.
J.

# Steps to reproduce

normal use.

# FreeCAD Info

```
OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22756 (Git) AppImage
Build type: Release
Branch: master
Hash: ce87f586ee36ffe8ebcdcd3947ced4ab389ae795
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: Ubuntu 18.04**
* **Updated:** 1/24/2022

0penBrain, 2022-02-07:
@luzpaz At least with migration I saw you assigned this to me. :) OK, will try to take care of this. ;)

0penBrain, 2022-07-06:
@luzpaz I think nothing changed yet. Will keep updated in coming days.

0penBrain, 2022-07-12:
@luzpaz Actually this one is funny. I guess it's all about UI at the end : 
[image]

IMO the problem in `[mm]' that makes user think it's only mm. But actually you can enter any quantity (from my test).
I took an arbitrary picture and scaled its whole length to `20 in` and after that it is ... 508 mm. Perfect.

If someone can confirm this doing same kind of test, I think we can just change `[mm]' to `[default = mm]` in the UI and we are done.

Will try to confirm this also by looking at the code.

0penBrain, 2022-07-12:
Confirmed in code, the text is parsed as quantity, so there should be no problem using imperial units
https://github.com/FreeCAD/FreeCAD/blob/ff802c7274d6f60074ee2c1faa16a856a9e6c3bb/src/Mod/Image/ImageTools/_CommandImageScaling.py#L146

luzpaz, 2022-07-12:
Sweet, maybe we should remove the `[mm]` or notate in the UI about it being 'unit agnostic' ?

0penBrain, 2022-07-12:
OK, we have 2 possibilities. Both will adapt to user units when dialog is opened, and will only authorize length quantities.

1) Using InputField
Looks like this 
[image]

Advantages : has the "checker" at the right of the widget, and is the "standard' in the Python WB
Drawbacks : no mechanism to easily select only the numbers (while not selecting the unit), and default value is always "mm" if no unit is set

2) Using QuantitySpinBox
Looks like
[image]
Advantages : mechanism to select only number (without unit) -- but not really usable in this case --, and default unit (if no unit entered) is user unit (eg. inches for imperial)
Drawbacks : Rarely seen in FC Python WB dialogs, so might look a bit inconsistent

Let me know what is your preference (if any)

luzpaz, 2022-07-12:
Probably the 1st option for consistency, right?

0penBrain, 2023-02-11:
@donovaly Can we close ?

donovaly, 2023-02-11:
> @donovaly Can we close ?

The bug is not fixed in 0.20 branch, right?
The idea is to keep bugs open that are not fixed in the stable branch so that users who are affected can find them as already known. They will also get the info that it is already fixed for the next major release, only have to be patient. Then they can report back to e.g. force us to backport in case we simply forgot about it. Hence there is the  "Fixed for 0.21" label in GitHub.

0penBrain, 2023-02-15:
> The idea is to keep bugs open that are not fixed in the stable branch so that users who are affected can find them as already known. They will also get the info that it is already fixed for the next [...]

It means that when we tag a commit with something like `fixes #xxxx`, that will automatically close the issue on merging, but then we have to reopen the issue, add a label to it, and finally later think about closing it.
This doesn't look like a friendly workflow. :confused:

donovaly, 2023-02-15:
> This doesn't look like a friendly workflow. 😕

I think it is (is also my experience in 10 years with the lyx.org project). Let's compare both workflows:

Method A: Keeping existing bugs open until fixed in stable:
- for the developer it is one click to reopen the bug and some seconds to set the "Fixed for 0.21" label
- for the release manager it is just a dozen clicks to close about the dozen issues
- forum guys only have to be active once a user reports it, but can then quickly find the open issue and post this. In many cases users will not even post to the forum if they found the existing bug. So let's say for every user encountering the issue, half will find the open issue, so no work at all for the forum guys. For the other half, we need about 10 minutes each

Method B: keeping bugs closed, no matter if already available in stable release
- most users will post in the forum, so a lot of work for the forum guys to read the report, then to check open _and_ closed bugs, then maybe a short discussion why the issue is closed despite still in stable...

So in total more work for us with method B while. We used in the project www.lyx.org for more than 10 years method A and it worked well. (that time I was very active in the forum and therefore know how much time it can take to give support. This sums up over the year.)

Roy-043, 2023-07-16:
Closing as fixed.

commit 38473f24 (2022-07-25): Make image scaling command aware of user units (#7222)

* Image: image scaling command is user unit aware ; fixes #6118

- if unitless value entered, InputField uses user choosen base unit
 Before change, always used FC core base unit
 Only applies to GUI user entered values, value changed programmatically still ...
 ... considered in core base unit if applied unitless
closed 2023-07-16 by Roy-043