FreeCAD/FreeCAD #6119: Crash on loft with twisted circles.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4470

* **Reporter:** BruceL
* **Date submitted:** 10/19/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** 

# Original report text

When executing a loft with a circles with 'twisted' starting points. FreeCAD will crash.

# Additional information

```FreeCAD 0.19, Libs: 0.19R22665 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Loading Assembly4 WorkBench
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3be30) 
#1  /tmp/.mount_FreeCAA40B2C/usr/lib/./libTKBool.so.7(+0x270559) 
#2  0x7fd54f9aef1b in BRepFill_CompatibleWires::SameNumberByPolarMethod(bool) from /tmp/.mount_FreeCAA40B2C/usr/lib/./libTKBool.so.7+0x130b
#3  0x7fd54f9b487a in BRepFill_CompatibleWires::Perform(bool) from /tmp/.mount_FreeCAA40B2C/usr/lib/./libTKBool.so.7+0xb6a
#4  0x7fd554b3b196 in BRepOffsetAPI_ThruSections::Build() from /tmp/.mount_FreeCAA40B2C/usr/lib/./libTKOffset.so.7+0x296
#5  0x7fd4ec437037 in PartDesign::Loft::execute() from /tmp/.mount_FreeCAA40B2C/usr/lib/_PartDesign.so+0x7f7
#6  0x7fd578c66ea2 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADApp.so+0xf2
#7  0x7fd5552a713c in Part::Feature::recompute() from /tmp/.mount_FreeCAA40B2C/usr/lib/Part.so+0xc
#8  0x7fd578bf906c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADApp.so+0x31c
#9  0x7fd578c528bc in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADApp.so+0x60c
#10  0x7fd578cba5b2 in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADApp.so+0x382
#11  0x7fd578cbaa54 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADApp.so+0x24
#12  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(PyCFunction_Call+0xf7) 
#13  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x250) 
#14  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4ea2) 
#15  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x2e2) 
#16  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x44) 
#17  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCode+0x1c) 
#18  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(+0x219ecf) 
#19  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(+0x252c44) 
#20  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libpython3.8.so.1.0(PyRun_StringFlags+0x7d) 
#21  0x7fd57898a11a in Base::InterpreterSingleton::runString(char const*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADBase.so+0x6a
#22  0x7fd579723af8 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADGui.so+0x98
#23  0x7fd4d1f97bd2 in void Gui::_cmdDocument<char const (&) >(Gui::Command::DoCmd_Type, App::Document const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const (&) ) from /tmp/.mount_FreeCAA40B2C/usr/lib/PartDesignGui.so+0x4b2
#24  0x7fd4d1f98465 in PartDesignGui::TaskDlgFeatureParameters::accept() from /tmp/.mount_FreeCAA40B2C/usr/lib/PartDesignGui.so+0x145
#25  0x7fd4d1f9be98 in PartDesignGui::TaskDlgSketchBasedParameters::accept() from /tmp/.mount_FreeCAA40B2C/usr/lib/PartDesignGui.so+0x4b8
#26  0x7fd4d2028259 in PartDesignGui::TaskDlgLoftParameters::accept() from /tmp/.mount_FreeCAA40B2C/usr/lib/PartDesignGui.so+0x579
#27  0x7fd5799c44e9 in Gui::TaskView::TaskView::accept() from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADGui.so+0x59
#28  0x7fd5773c60d2 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Core.so.5+0x780
#29  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5(+0x274d73) 
#30  0x7fd5773c60d2 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Core.so.5+0x780
#31  0x7fd577c73360 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5+0x32
#32  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5(+0x20da19) 
#33  /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5(+0x20e2ae) 
#34  0x7fd577c7434a in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5+0x88
#35  0x7fd577c05c23 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5+0x135
#36  0x7fd577bda83a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5+0xde
#37  0x7fd577bdf0a9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5+0x79d
#38  0x7fd5796f10c9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libFreeCADGui.so+0x89
#39  0x7fd5773afa9c in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Core.so.5+0x88
#40  0x7fd577bde553 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAA40B2C/usr/bin/../lib/libQt5Widgets.so.5+0x283
[... 30 lines cut ...]
[forum post](https://forum.freecadweb.org/viewtopic.php?f=3&t=51073&p=438938#p438938)

# FreeCAD Info

```
OS: Debian GNU/Linux bullseye/sid (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22665 (Git) AppImage
Build type: Release
Branch: master
Hash: b0d7850406e046f3b7a8cd5da261e7d54df6bd99
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/Canada (en_CA)
```

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

FEA-eng, 2024-10-08:
I can't reproduce it. I guess it was fixed by the commit mentioned here?

[part3_new.FCStd.zip](https://github.com/user-attachments/files/17298208/part3_new.FCStd.zip)

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

luzpaz, 2024-10-08:
> What's our procedure for bugs that are fixed in the occt fork? If left open it might be worth adding a tag like bloblfish-fixed to know it has been adressed but not upstreamed.

@adrianinsaval you wrote this in the forum thread. Any progress on how we deal with this ?

adrianinsaval, 2024-10-08:
I don't know, @maxwxyz how do you think we should deal with things like this? We have a fix for this on our fork for occt but it's not in upstream occt.

regarding my previous suggestion, `blobfish-fixed` is not appropriate anymore as we have our own fork now, not directly blobfish's (although a majority of the patches still come from him)

maxwxyz, 2024-10-09:
@adrianinsaval will it be merged upstream and is something to be done in FreeCAD if merged upstream? If there is no action in this repo I'd just close the issue as completed. IMO the upstream PR (or if there is an issue in the OCC tracker) is enough.

closed 2024-10-09 by maxwxyz

luzpaz, 2024-10-09:
re: OCCT   
We may be able to convince @dpasukhi to support the upstream patching effort ? :thought_balloon:

dpasukhi, 2024-10-09:
Hello, I was trying to connect with blobfish to propagate his patches.
I still have no response. I can't just merge them without approve into official repo.

Only one option - it is patches with closest idea. His patches are not stable. I plan to work on them to rework them and optimize.
But I still prefer to have CLA :(
For sure we are working on simplifying CLA process, but for France it is not so easy.

luzpaz, 2024-10-09:
cc @blobfish

blobfish, 2024-10-09:
> Hello, I was trying to connect with blobfish to propagate his patches. I still have no response. I can't just merge them without approve into official repo.
> 
> Only one option - it is patches with closest idea. His patches can be improved a little and new tests can be added. I plan to work on them to rework them and optimize. But I still prefer to have CLA :( For sure we are working on simplifying CLA process, but for France it is not so easy.

> cc @blobfish

I think I have been consistent and clear. I have no intention of signing anybodies CLA. I don't care how easy the process gets.

dpasukhi, 2024-10-09:
Thank you. I pure understand you.
In that case I can't merge changes.
@luzpaz could you share critical bugs? I can try to resolve in our way.

luzpaz, 2024-10-11:
@adrianinsaval can we get the critical bug to dpasukhi ?

adrianinsaval, 2024-10-11:
Unfortunately I don't know how to make the DRAW script thing to reproduce in plain occt, it doesn't look like we reported this to occt properly