FreeCAD/FreeCAD #6120: Place and orientate dimension constraints in one go

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4472

* **Reporter:** Sandkuchen
* **Date submitted:** 10/23/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Concerned environment/workbench: Sketcher

Requested Feature: Ability to orientate the labels/arrows of dimensional constraints (aka drag them to a convenient place within the the sketch) while still being within the constraint command

The feature is described and to some degree discussed in https://forum.freecadweb.org/viewtopic.php?f=8&t=51125

The requested feature is not related to the consolidated dimension constraint tool (aka "Smart Dimension" etc.) described in https://forum.freecadweb.org/viewtopic.php?f=8&t=50521 apart from both requests dealing with dimensional constraints in general. Neither feature depends on the other; neither includes the other; neither makes the other obsolete; they don't come as a set.

# FreeCAD Info

```
OS: Windows 7 SP 1 (6.1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22665 (Git)
Build type: Release
Branch: master
Hash: b0d7850406e046f3b7a8cd5da261e7d54df6bd99
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: German/Germany (de_DE)

```

# Other bug information

* **Priority:** low
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2022-04-03:
@paddlestroke this looks like something you've been working on, right?

PaddleStroke, 2022-04-08:
Dev branch : https://www.youtube.com/watch?v=J7khOBhch44&t=53s&ab_channel=FreeCADDevDiary
Video showing the tool : 
https://www.youtube.com/watch?v=J7khOBhch44

luzpaz, 2022-04-08:
Thanks! 
Can you also add the forum thread to the above reply if you don't mind (so users can follow the development discussion)

maxwxyz, 2024-04-18:
@PaddleStroke can this be closed now?

PaddleStroke, 2024-04-18:
Yes

closed 2024-04-18 by maxwxyz