FreeCAD/FreeCAD #6122: Path Inspect tool doesn't respect user's preferred units

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4476

* **Reporter:** sliptonic
* **Date submitted:** 10/27/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Inspecting gcode always shows in FreeCAD internal units (mm) rather than the user's selected unit.

Make the inspection tool smarter about units.

# FreeCAD Info

```
OS: Linux Mint 19.3 (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22865 +9 (Git)
Build type: Unknown
Branch: feature/toolbitdock
Hash: 0a3ff2279f99b5b15925eb03aff5ab1567b1a798
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2022-09-11:
Related to 'units' like #6118

sliptonic, 2023-08-12:
closing.  Inspect shows the path commands.  Not a bug

closed 2023-08-12 by sliptonic