FreeCAD/FreeCAD #6125: ShapeBinder can or not be padded depending on pad type

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4484

* **Reporter:** openBrain
* **Date submitted:** 11/3/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

When trying to pad a ShapeBinder :
* If pad type is "Dimension", I can well pad a ShapeBinder.
* If pad type is "To first", FreeCAD rejects it with the error message "Linked object is not a Sketch or Part2DObject".

[Forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=51798)

# Steps to reproduce

Cf. forum

# FreeCAD Info

```
OS: Ubuntu 18.04.5 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22894
Build type: Release
Branch: master
Hash: 9eb080488d970d313c538473e7272117ea0a7cd1
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

shyblower, 2022-03-09:
The same issue applies to pockets based on ShapeBinders.
It works with Realthunder's Link Daily release 2022.01.11, though.

maxwxyz, 2024-08-28:
is this issue still relevant?

FEA-eng, 2024-10-08:
> is this issue still relevant?

I can't confirm it. Looks fixed but someone else should check just in case.

[ShapeBinder_issue_1_0.FCStd.zip](https://github.com/user-attachments/files/17298107/ShapeBinder_issue_1_0.FCStd.zip)

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

maxwxyz, 2024-10-08:
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

closed 2024-10-08 by maxwxyz