FreeCAD/FreeCAD #6127: TechDraw - normative correct simplified presentation of Thread in Top View and Cross section according to ISO 6410–1 standard

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4489

* **Reporter:** maker
* **Date submitted:** 11/10/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

There is actually no feature in Techdraw to paint a normative correct simplified presentation of Thread in Top View and Cross section according to ISO 6410–1 standard. 
Solution approaches:
1) TechDraw 
According to the forum thread https://forum.freecadweb.org/viewtopic.php?f=35&t=46183&hilit=Threads#p397091 there are first efforts put in a macro for the Top View:
https://github.com/reox/FreeCAD_macros/blob/master/TDThread.FCMacro
2) PartDesign
Another proposal could be probably realised with the PartDesign Hole feature. See the post:
https://forum.freecadweb.org/viewtopic.php?f=35&t=51236&hilit=Threads&start=10#p449851

# Additional information

https://wiki.freecadweb.org/Wished_tools#2D drawing

Thread - PartDesign workbench. Feature Hole: correct functionality of bore depth according to ISO 6410–1 standard
Thread - normative correct simplified presentation of cross section in Techdraw according to ISO 6410–1 standard

# Steps to reproduce

According to the following forum threads:

Threads in TechDraw
https://forum.freecadweb.org/viewtopic.php?f=35&t=51236
Threading Visualization in TechDraw
https://forum.freecadweb.org/viewtopic.php?t=44319
Feature Request: Threads
https://forum.freecadweb.org/viewtopic. ... 5&t=27376

# FreeCAD Info

```
OS: openSUSE Leap 15.2 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Unknown
Build type: Release
Python version: 3.6.10
Qt version: 5.12.7
Coin version: 4.0.0
OCC version: 7.5.0
Locale: German/Germany (de_DE)
```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: openSUSE Leap 15.2**
* **Platform:** PC
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by maker 2020-12-05 22:03
Allocation to part design workbench preferred. See addition to description dated 2020-12-05.
Should be reviewed in extension and in comparison to existing ticket https://tracker.freecadweb.org/view.php?id=3369 dated 2018-03-10.

WandererFan, 2024-01-29:
additional information from @domad
[Suggestion schematic-technical representation of threads .pdf](https://github.com/FreeCAD/FreeCAD/files/14085519/Suggestion.schematic-technical.representation.of.threads.pdf)

pierreporte, 2025-04-29:
A **200 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.

franklincg, 2026-08-17:
Hi, I'm interested in working on this bounty. Could you confirm the current reward amount and whether the issue is still available for a new contributor to work on?