FreeCAD/FreeCAD #6129: In French version Preferences vanish on MacOS

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4491

* **Reporter:** chrisb
* **Date submitted:** 11/13/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** feedback
* **Tags:** #pending, locale, macOS, translation, UI

# Original report text

If language is changed to Français on MacOS, then the preferences no longer open the preferences dialog.
As a workaround they are available from the Start page.
See [forum posts](https://forum.freecadweb.org/viewtopic.php?f=34&t=51942)
and https://forum.freecadweb.org/viewtopic.php?f=12&t=51624

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22960 (Git)
Build type: Release
Branch: master
Hash: c5a4b01d2e4218bcc0eb6650337650a6c65ef0e4
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: MacOS **
* **Updated:** 1/23/2022

# Discussion from Mantis ticket

### Comment by chrisb 2021-07-26 07:39
Same issue still present with Russian locale, see [forum discussion](https://forum.freecadweb.org/viewtopic.php?f=8&t=60559).

---

### Comment by Kunda1 2022-01-23 00:52
chrisb is this still an issue?

luzpaz, 2022-03-15:
@yorikvanhavre is chrisb on github ?

PLChris, 2022-03-16:
> @yorikvanhavre is chrisb on github ?

Yes, I am.

PLChris, 2022-03-16:
This issue is fixed. However, I still cannot close it - or I don't know how to.

luzpaz, 2022-03-16:
When @yorikvanhavre gives you permissions you'll see the following 2 buttons at the bottom of the page
[image]

PLChris, 2022-03-17:
Retested in 0.20. Issue doesn't occur anymore, French preferences are shown correctly on macOS.

closed 2022-03-17 by PLChris