FreeCAD/FreeCAD #6130: Loft between sketches with inner and outer elements created in different order results in bad 3D geometry

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4498

* **Reporter:** robotarmy
* **Date submitted:** 11/21/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** #pending-forum, Loft

# Original report text

When creating a loft, the order the sketch elements are created should not matter, but it does. 

Take the case of lofting between two sketches, with two circles each, for example, to make a kind of cylinder. In one sketch the inside was created first, and in the other sketch the outside was created first, the loft tries to cross over itself resulting in zero thickness geometry and a 'black part'. 

See forum post https://forum.freecadweb.org/viewtopic.php?p=449244#p449244
See updated limitations on wiki page for PartDesign Loft: https://wiki.freecadweb.org/index.php?title=PartDesign_AdditiveLoft&type=revision&diff=794134&oldid=763800

Thanks!

# Steps to reproduce

See attached file

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22756 (Git)
Build type: Release
Branch: master
Hash: ce87f586ee36ffe8ebcdcd3947ced4ab389ae795
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)
```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Bug
* **OS: Mac OS 10.15.7 (19H2)**
* **Platform:** x86
* **Updated:** 2/6/2021

brandonbeaumia, 2022-06-19:
@donovaly I just saw the work you did for 0.20 to get nested profiles working in Part Design (much appreciated!). Is it possible that the work you've already done also brought this bug closer to a fix or is that wishful thinking?

donovaly, 2022-06-19:
I tested and when you take the loft and make a new one using the two sketches, you get a correct loft.

There is a bug in OpenCascade regarding circles. Therefore in FreeCAD 0.19 the detection what is the inner one failed. I fixed this for FreeCAD 0.20 at least for PartDesign Pad/Pocket and Part Extrude.

If there is still an issue with lofts in FreeCAD 0.20, please open a new forum thread with a test file and CC me there..

brandonbeaumia, 2022-06-20:
Hm...I just tested both the file from the original post (vacuumadapter.fcstd), and also started from scratch on a new file using both rectangles and circles. The behavior is still not working correctly that I can tell: if you don't create the pieces in the same order from one sketch to the next, your faces will cross. I've attached my new from-scratch file plus screenshots from both it and the original discussion's file showing the crossing faces.

I can learn the whole bug report process and start this report over back on the forum if you'd like (I'm willing!), I just don't want to bug everyone with a duplicate if everything here still applies like it seems to.

[image]

[image]

[Example bad lofts.zip (change to .fcstd)](https://github.com/FreeCAD/FreeCAD/files/8936718/Example.bad.lofts.zip)

donovaly, 2022-06-20:
I could reproduce the issue and opened a new forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=69610

I remember a cool PR from @mwganson that would resolve the issue too.

gryphonB-2, 2024-11-23:
When creating some kind pipe, this issue still exists in RC4.
That's regardless of the shapes used as a basis for the loft, as I found this morning.
When creating a pipe the order of creation of the inner & outer cross sections must be the same for all sections. 
I've always thought this was an easy fix: either calculate the cross section areas or just the delta x & delta y of the two cross sections in each sketch, and use those values to specify the inner & outer cross sections. 
Or just include a check box to flip inner & outer, so the walls do not end up crossing. 
There's no example file or image, the above say it all.

mwganson, 2024-11-23:
I don't know about cool, but it was colorful at least.  That PR was for additive lofts, I think, but really it should be done at the sketch level instead.  A simple reverse wires boolean property could work where there are only 2 wires involved, but in a more complex sketch, for example a 5 hole pattern inside a circle, then it becomes much more complicated to reorder the wires.  It makes for a difficult UI because learning the UI is more difficult than simply redoing the sketch, and typically once the user learns that the wire creation order is critical, then the need to reorder the wires tends to go away.

I did tackle this issue with a macro (WireFilter) and in MeshRemodel workbench the SketchPlus object has a wire filter option in it, among many other options for sketches, such as parametric arrays, scaling, and offsetting.  If you select an existing sketch and create the SketchPlus with that sketch selected it will basically make a simple non-parametric copy of the existing sketch (including all constraints and references to external geometry), and then you can use the SketchPlus in place of the original.

brandonbeaumia, 2025-02-23:
I can confirm that this issue still exists in newest weekly build 40317 and still frustrates me on a semi-regular basis (at least for more complicated sketches).

brandonbeaumia, 2025-05-27:
Build 41694 still has the issue.

freckletonj, 2026-08-27:
I like @PeonyThePeon 's suggestion ^^^

I'm not clear, is this issue because of the winding of wires *as they're drawn* or when they're *constrained*? Knowing that would help us debug specific instances - I just faced this issue too (plz forgive the duplicated issue): https://github.com/FreeCAD/FreeCAD/issues/32215

Some simple low risk proposals that would also resolve this:

1. we have a way to inspect winding orders in sketcher, or overlaid on SubShape Bindings. This could be a simple directional arrow on wires, a visual we can toggle on/off, one arrow for each wire
2. or we can inspect winding ordering by simply coloring things (in a toggleable way) with eg red means CW winding and blue means CCW. I think this binary is how it eventually gets interpreted, so we may not need per-edge diagnostics if we only need per-face
3. we have a way to flip the winding, or orderings in the sketcher and/or SSBs (eg changing directionality of edges, or reordering constraints, or flipping whole regions of connected wires)
4. at Loft time, we have a way of reordering or flipping wires we've selected. This would be nice in cases where the same geometry is a member of multiple different lofts (etc), possibly with different or incompatible windings.

codeotaku, 2026-08-30:
I'm looking into this, since lofts may be [getting an update](https://github.com/Open-Cascade-SAS/OCCT/pull/1506) soon.

I don't think it's a winding issue (there is another occ method that kind of corrects/adjusts for that, and it would produce quite different looking results if/when it's wrong).  There is already a method `FaceMakerCheese::isInside` in FreeCAD to check whether one wire is inside another... which, if we use to reorder what we send to OCC has the desired effect:

[image]

[image]

Basically, my understanding of this issue is that FreeCAD extracted each section’s wires in "raw" sketch creation order (by index), and this connected outer→inner and inner→outer, creating intersecting loft surfaces. 

@pieterhijma OCCT built the loft that FreeCAD requested, so not OCCT.

codeotaku, 2026-08-30:
Also from [this duplicate issue](https://github.com/FreeCAD/FreeCAD/issues/32215)

before: 

[image]

after:

[image]

Created regression tests from both of these issues.

closed 2026-09-12 by kadet1090