FreeCAD/FreeCAD #6131: Can't save color

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4499

* **Reporter:** johnwang
* **Date submitted:** 11/22/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Talked in the post:
https://forum.freecadweb.org/viewtopic.php?f=3&t=52341

Using this code to generate a FemMesh object with two triangle elements. Then changed the color of one element. But after saving and reopen, the color is lost.

# Steps to reproduce

Use this code to create the file, then save and reopen.
```import Fem

a = Fem.FemMesh()

a.addNode(0,    0,    0,    1)
a.addNode(10,    0,    0,    2)
a.addNode(5,    10,    0,    3)
a.addNode(15,    15,    0,    4)

a.addFace([1,2,3], 1)
a.addFace([2,4,3], 2)
obj = FreeCAD.ActiveDocument.addObject("Fem::FemMeshObject")
#obj.Placement.Base = FreeCAD.Vector(2, 0, 0)
obj.FemMesh = a

obj.ViewObject.ElementColor = {1:(1,0,0)}```

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22900 (Git)
Build type: Release
Branch: master
Hash: b7b4250b7fe5e98bcdbe2a7c7be256132b35cd3d
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Windows 10 **
* **Updated:** 2/4/2022

wperkins, 2022-03-11:
I also experience this with meshes created by the Mesh WB. Recreate as follows:

1. With Part WB, create a cube. 
2. Right click on cube and "Set Colors..."
3. Color some cube faces.
4. With the Mesh WB, "Create mesh from shape". A colorful mesh is created.  This can be exported as *.obj and colors are correct. 
5. Save and close project.
6. Reopen project and  the mesh is all gray.

The attached was created with Revision 8204fc84. 
[mesh-color-test.zip](https://github.com/FreeCAD/FreeCAD/files/8235205/mesh-color-test.zip)

luzpaz, 2022-03-11:
@wperkins please edit the above comment and add your full [About](https://wiki.freecad.org/About) info. Thanks

luzpaz, 2022-04-13:
Confirmed on 
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28690 (Git)
Build type: Release
Branch: master
Hash: 5ab7d17f67969b12c53b7bb8715f198093fcf56b
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * pinger
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * Curves 0.4.2
  * Ship 778610208.0.0
  * BIM 2021.12.0
  * A2plus 0.4.56
```

luzpaz, 2022-05-03:
Can confirm on 
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28786 (Git)
Build type: Release
Branch: master
Hash: ccc8ac65c5a04f7f6690c29dc08225216f201e85
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
```

luzpaz, 2022-07-08:
> Mesh and FEM are two separated and independent wb but the problem exists in both of them. The commit handles it for the former and for the latter it still must be implemented.

@wwmayer shall I close this ticket and open a separate ticket for the FEM portion of the fix?

luzpaz, 2022-07-26:
@berniev any progress on this?

berniev, 2022-07-26:
Not one I'm working on. The commit was in my fork and likely an accident. But how does that even show up in the main repo?

wwmayer, 2022-07-27:
> @wwmayer shall I close this ticket and open a separate ticket for the FEM portion of the fix?

It's fine as it is.

luzpaz, 2023-02-11:
@coreyoconnor does your branch fix this issue?

coreyoconnor, 2023-02-11:
@luzpaz unfortunately - my branch is just a merge of @sliptonic 's branch. I do not know if the change impacts this issue.

wwmayer, 2024-03-13:
It's not a show stopper issue. So, doing it after 1.0 shouldn't be a problem.

PR #13698 (2024-05-06): Fem: Save and restore custom colors of elements in mesh - fixes #6131
This fixes #6131 FemMesh coloring issue (fix #5815 requires one more step, since currently only first-order meshes are colored correctly).
With this pull request it is possible to save and restore custom colors on meshes, including transparency.
An enumeration property `ColorMode` is added to select the color mode. There are three options:
 * `Overall`: The color of the object is determined entirely by `ShapeAppearance` property.
 * `ByElement`: The mesh is colored by element based on the values of the `ElementColorArray` pr [... cut]
closed 2024-05-06 by yorikvanhavre