FreeCAD/FreeCAD #6133: Dimension between 2 concentric circles

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4503

* **Reporter:** Mistermaker
* **Date submitted:** 11/25/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** constraints, sketcher

# Original report text

Request is a method on how to add a dimensional constrain between 2 concentric circles. This is closely related to the requested [offset feature](https://tracker.freecadweb.org/view.php?id=3708). See added gif for how it is currently done in Fusion360 and how it is done in Freecan 0.19

See forum topic:
https://forum.freecadweb.org/viewtopic.php?p=450998#p450998

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22960 (Git)
Build type: Release
Branch: master
Hash: c5a4b01d2e4218bcc0eb6650337650a6c65ef0e4
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Dutch/Netherlands (nl_NL)

```

# Other bug information

* **Priority:** none
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

adrianinsaval, 2022-02-14:
#5439 is trying to implement this

adrianinsaval, 2023-03-28:
issue can be closed, should be solved by #8896 and #8967 

@PaddleStroke consider using this new constraint when/if you resume work on the offset feature

closed 2023-03-29 by luzpaz

PaddleStroke, 2023-03-29:
Nor sure it will be useful for offset, but for constrain contextually it will. I'll definitely go back to these as soon as abdullah is ready to start merging tool settings.