FreeCAD/FreeCAD #6136: Grouping problem (object disappearing after drag drop into group)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4508

* **Reporter:** hardrock
* **Date submitted:** 12/9/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** #tobeclosed

# Original report text

If group and object has same prefix, then when putting object into this group, or pulling out, object may dissapear :)
After close / open project, dissapeared object will be on place before grouping operation, even if project was saved after object disappearing.

# Additional information

https://forum.freecadweb.org/viewtopic.php?t=51344

# Steps to reproduce

1. Create a new group for example with name "Test_Group".
2. Create any kind of object, for example cube with name "Test_Cube". Name start should be the same with name start of the group.
3. Drop in / Drop out box into  / from group untill it dissapear from project tree.

Object may dissapear after first drop, or may be after few.

# FreeCAD Info

```

OS: Ubuntu 18.04.4 LTS (Unity:Unity7:ubuntu/unity)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 983e513704a659c1008fbaf95da6a34975cc37ac
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Russian/Russia (ru_RU)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 18.04**
* **Platform:** x86
* **Updated:** 1/25/2022

# Discussion from Mantis ticket

### Comment by hardrock 2020-12-10 08:50
Seems like it happens not only when names have same prefix. In fact name doesn't matter.

---

### Comment by chennes 2021-02-07 03:56
I could not reproduce on:
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23935 (Git)
Build type: Debug
Branch: master
Hash: d5b68febbb4989b290b9af36bb2cb21d61a87a2e
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

---

### Comment by Kunda1 2021-02-20 13:37
hardrock please retest on latest v0.19_pre 

Edit:
Can't reproduce on 
OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24054 (Git) AppImage
Build type: Release
Branch: master
Hash: ed873038025277f220dfaa058599a0dfbb435835
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

---

### Comment by hardrock 2021-02-21 15:40
New video.

https://gofile.io/d/zL8ZUN

Try to drop in drop out multiple times. Problem appears not always, but frequently.

OS: Ubuntu 18.04.4 LTS (Unity:Unity7:ubuntu/unity)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 3d227cf7179c00e57f2733f947b4b1b0368d219a
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Russian/Russia (ru_RU)

---

### Comment by Kunda1 2022-01-25 18:57
hardrock the clip is missing. Can you re-upload?

adrianinsaval, 2022-02-10:
nobody besides OP was able to reproduce this, should this be kept?

closed 2022-05-25 by luzpaz

Julrob199, 2025-11-12:
I still have a similar problem... When I drag and drop a group in the tree view (maybe drop it on itself), it just disappears.

OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.0.2.39319 (Git)
Build type: Release
Branch: makepkg
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.13.7, Qt 6.9.2, Coin 4.0.5, Vtk 9.5.1, OCC 7.9.1
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * A2plus
  * Assembly4.backup1760093069.2768822 (Disabled)
  * Assembly4 0.60.2
  * Render 2024.12.15