FreeCAD/FreeCAD #6139: Alternative constructions methods for B-Spline

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4512

* **Reporter:** abdullah
* **Date submitted:** 12/16/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Look at this post from Werner:
https://forum.freecadweb.org/viewtopic.php?f=10&t=51716&start=60#p456903

Look at this user example of DIY poly-bezier:
https://forum.freecadweb.org/viewtopic.php?p=460270#p460270

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

maxwxyz, 2024-02-21:
@AjinkyaDahale is this solved with bezier curves?

AjinkyaDahale, 2024-02-22:
> @AjinkyaDahale is this solved with bezier curves?

Bezier curves are not currently implemented. However, splines-through-points are. The issue is unclear on what method it asks for.

maxwxyz, 2024-02-22:
I see but with your implementation of bezier curves this could ultimately be closed?

AjinkyaDahale, 2024-02-24:
> I see but with your implementation of bezier curves this could ultimately be closed?

Again, depends on how you define it. There is already an "alternative method" but it doesn't make the hermite splines as mentioned in the post by wmayer. They don't exist in OCCT already, so adding Bezier will not solve it either.

I would suggest one of two things:
1. Close this as being too ambiguous.
2. Rename this to the lines of "add Cubic Hermite splines". If similar issue already exists, this is moot and just close it.