FreeCAD/FreeCAD #6140: Arch Space, showing wrong "Line Spacing" in 3d-View

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4516

* **Reporter:** balrobs
* **Date submitted:** 12/19/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Changing the view-properties "First Line" or "Font Size" of an Arch Space object leads to correct text heights rappresentations but not to the correct "Line spacing" in the 3d-View. To achieve this it's neccessary to reset the view-property "Line spacing" too.
See forum topic: https://forum.freecadweb.org/viewtopic.php?f=23&t=53364

# Additional information

Expected behaviour: Changing view-properties "First Line" or "Font Size" without changing "Line Spacing" should be enough to get a correct rapresentation in the 3d-View.

# Steps to reproduce

1. Create Arch Space with Arch WB
2. Set Space view-properties to "First Line=200mm", "Font Size=200mm", "Line Spacing=1"
3. 3d-View changes consequently
4. Change Space view-properties to "First Line=400mm", "Font Size=400mm"
5. 3d-View changes, but "Room label" and "Room area" are shown vertivally overlaped
6. To correct the 3d-View the view-property "Line Spacing" must be changed e.g to another value then 1 and then reset to 1

# FreeCAD Info

```
OS: Ubuntu 20.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 5cea165a6691a3dc7d0830510115fbf50b81ee00
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.3.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

luzpaz, 2022-05-25:
@WandererFan can you test this in TechDraw and confirm?

closed 2022-12-01 by donovaly

balrobs, 2022-12-01:
I confirm the issue is gone.
Thanks for solving and merging!

PR #7922 (2022-12-01): Arch: Fix Arch_Space text position
Fixes #6140.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---