FreeCAD/FreeCAD #6150: Export GCode - 'PrimitivePy' object has no attribute 'Proxy' error message

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4537

* **Reporter:** bertrand.lancelot@objectspirit.net
* **Date submitted:** 1/17/2021
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

OS: MAC Catalina 10.15.7
Freecad version 018.16146 (same problem with install from a download or with brew install)

Impossible to export the 3D model to print it with my 3D printer. Anyway the processor used.
Then this issue is blocking issue for me.

Thanks in advance, but in waiting, I will try another tool to be able to test and use my new 3D printer.

# Steps to reproduce

create a symple Cylinder 3D object or anything else in 3D.
Go to menu file / export
    title: test.gc for exemple
    validate
    select the first processor in the list "centroid_post" (none is working after many tests)
    Validate

Then the error message appear:
Cylinder
postprocessing...
Traceback (most recent call last):
  File "<string>", line 4, in <module>
<class 'RuntimeError'>: 'PrimitivePy' object has no attribute 'Proxy'

# Other bug information

* **Priority:** high
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by bertrand.lancelot@objectspirit.net 2021-01-24 08:02
Hi,

I did find a turnaround in using ultimaker cuba tool. Then, I have to export to obl file and open it in cuba to transform obj file to g-code.
A bit longer but acceptable in waiting a fix.

So, possible to change priority to low if you want.
Regards,

luzpaz, 2022-05-03:
@Russ4262 have you seen an issue like this before?

Russ4262, 2022-05-03:
Reading the Original Report Text, I see, "Impossible to export the 3D model to print it with my 3D printer."  This has nothing to do with Path.  As for what the reporter is wanting to do, export a solid for printing with a 3D printer, this has been functioning successfully since at least 0.19, if not previously.  The shape export format for 3D printer slicing is not ".gc" as far as I know.  I use Cura Ultimaker for my slicing needs, and I use FreeCAD to export solids on a regular basis when doing so, usually the ".stl" format is what I use personally, though other formats are accepted by my slicing software.

I think the reporter misunderstood what the export tool was capable of and intended to do.  If the reporter was expecting that exporting as a ".gc" file would auto-slice the solid and write that g-code to file, they were mistaken.

Another issue I see with this report is that the user reported using 0.18.16xxx.  That is an old version of FC now.  We are at 0.20.287xx, which is a lot of changes.

My opinion is we proceed with one of the following:
1.) We need the reporter to test with a recent weekly build and provide the feedback from that updated test. If I follow the instructions provided, I do not get the result stated in this report.
2.) Ask other devs to weigh in on the issue, as they might understand it better.
3.) Close this issue and mark it as fixed. I am not aware of another known issue for exporting g-code from a job, if that was the user's intention, and currently job post-processing is functioning.

My FC info for this round of tests.
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28751 (Git)
Build type: Release
Branch: master
Hash: d36cbb522ae91574b8ff41257653a514c1cb4beb
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.0
Locale: English/United States (en_US)
Installed mods: 
  * FC_SU
  * Z_MacroStartup
[/code]

luzpaz, 2022-05-04:
Thx for the comprehensive answer. Closing sounds like the right move

closed 2022-05-04 by luzpaz