FreeCAD/FreeCAD #6152: Add new method for trim sketch

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4544

* **Reporter:** Jee-Bee
* **Date submitted:** 1/25/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** sketcher, trim

# Original report text

Add a new method for trimming multiple sketch lines at once.
Description of work flow
- Model a shape that needs a trim
- start trim tool
- click at some point and while clicking move the mouse
- All lines that are crossed remove them till nearest intersection(s)
See images for simple example

# Additional information

see: https://forum.freecadweb.org/viewtopic.php?f=10&t=54441#p468692

# Other bug information

* **Priority:** low
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2022-04-03:
@abdullahtahiriyo shares his perspective:
> It involves having a new mode in the trim tool and changing what to do when moving the mouse and clicking.
>I was thinking about this behaviour of deleting when there is nothing left after trimming: 

[image]

luzpaz, 2022-04-03:
CC @paddlestroke

PR #13188 (2024-04-15): Sketcher: Trim tool, hold click and move. Fixes #6152
Sketcher: Trim tool, hold click and move. Fixes #6152

https://github.com/FreeCAD/FreeCAD/assets/19984177/a0677d48-e11a-4936-8a1e-d6636397d69a
commit 4b968a60 (2024-04-15): Sketcher: Trim tool, hold click and move. Fixes #6152
closed 2024-04-15 by yorikvanhavre