FreeCAD/FreeCAD #6156: Loft tool "Closed" option not working

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4554

* **Reporter:** GeoffG
* **Date submitted:** 2/7/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

I have created a model that contains five circle sketches arranged so that the solid can be closed using the Closed option. Unfortunately the Closed option does nothing in this case. The model file containing the sketches and resultant AdditiveLoft has been attached.

I have raised this in the forum and others have indicated that this is a bug. 

The forum topic is: https://forum.freecadweb.org/viewtopic.php?f=3&t=55253

# Additional information

I don't know if its useful but I have attached a screenshot of the versions of the graphical environment and processor that I'm using.

# Steps to reproduce

(I apologise if I'm being too descriptive for such a simple process)

1. Delete the existing AdditiveLoft
2. Select the 5 sketches in order from top to bottom.
3. Activate them by pressing the space bar
4. Switch to Part Design workbench
5. Initiate the Loft tool
6. Toggle on the Closed option

Nothing happens. There are no error messages either.

# FreeCAD Info

```
OS: Ubuntu 20.04.2 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23756 (Git) AppImage
Build type: Release
Branch: master
Hash: 9c6e9184930a52b165a0b7274e3a45d1006bfe67
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/Australia (en_AU)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Kubuntu 20.04**
* **Platform:** Linux
* **Updated:** 2/7/2021

adrianinsaval, 2022-02-15:
I've been taking a look at this and PD Loft is weird, the closed property is unused.
It uses BRepOffsetAPI_ThruSections which has a flag to make solids instead of shells, but PD generates shells then generates the faces of the start and end profiles in order to make a solid, this seems to be the reason why using the first profile as last section fails. I say this because Part workbench does use the solid flag and if the closed property is true it adds again the first profile as a last profile, except when the last user selected profile is a vertex, then it throws an error as this is not supported.

Anybody has any idea why PD does things like this?

Fat-Zer, 2023-03-05:
> Anybody has any idea why PD does things like this?

AFAIR that's done to support sketches with several wires i.e. creating pipe-like lofts. For each pair of wires on each section PartDesign creates a separate Loft surface, also a surface is created for the first and the last face. After that all those are getting sewn together into one solid.

I'll try to look into it... if BRepBuilderAPI_Sewing correctly handles sewing an object to itself, it should be a pretty trivial to fix...

Roy-043, 2023-07-16:
Closing as fixed.

closed 2023-07-16 by Roy-043